dronekit/dronekit-python #1070: dronekit doesn't change flight modes - DO_SET_MODE: UNSUPPORTED

EmirhanKarahan, 2020-09-28:
using dronekit-sitl, mavproxy and dronekit library. tried on ubuntu 16,18,20 and debian same problem.

im starting the code (simple_goto example) but the altitude doesn't change. if i check mavproxy terminal i see the error "DO_SET_MODE: UNSUPPORTED". if i change the mode stabilize to guided manually from mission planner, drone takes off and does the mission until the landing point. drone doesn't land because it's not able to change mode.

thank you

TulipDi, 2020-11-18:
I have the same error, have you solved it?

TulipDi, 2020-11-18:
> I have the same error, have you solved it?

I have solved it. The problem is in pymavlink.  
uninstall pymavlink with
`pip uninstall pymavlink`
and then reinstall the older version with
`pip install pymavlink==2.4.8`

EmirhanKarahan, 2020-11-18:
Thank you. I solved it with the same way too

closed 2020-11-18 by EmirhanKarahan

groers, 2021-12-08:
> > I have the same error, have you solved it?
> 
> I have solved it. The problem is in pymavlink. uninstall pymavlink with `pip uninstall pymavlink` and then reinstall the older version with `pip install pymavlink==2.4.8`

Thanks a lot. when I run the official example vehicle_state.py, I can't change the mode. I used your method to solve this problem.