dronekit/dronekit-python #1073: vehicle.message_factory.distance_sensor_encode not working?

Mytre, 2020-10-13:
I have my mission planner, set up OA_type to 1, set serial 1 to mavlink and I am connected through a pi, I am trying to now send obstacle data to my pixhawk for tes,t but for some reason sending this message:

    msg=vehicle.message_factory.distance_sensor_encode(
        0, #time since boot , this is not used
        5, #min distance in centimeters
        1800, #max distance of the sensor in cm
        x, #distance of object detection.
        0, #type 
        1, #onboard id, not used
        y, # sector
        0, # covariance, not used.
        )
    vehicle.send_mavlink(msg)

with values, is not triggering the obstacle detection, the proximity radar on mission planner is not showing any change. Is there a way to fix this or am I using the wrong type of message ?

Mytre, 2020-10-17:
Never mind, found that the issue was a wrong parameter set up.

closed 2020-10-17 by Mytre

suryadevelope, 2021-10-18:
could you please tell me what is the correct method to setup parameter you made, as if i used this method one of the arm is vibrating a lot

Dartherick, 2022-09-22:
> Never mind, found that the issue was a wrong parameter set up.

how did you solve it? what parameters did you change?