dronekit/dronekit-python #1081: RC controller mode changes not taking whne connected to onboard computer

lachlanbrewster1, 2020-11-25:
I am currently finding that my drone tends to ignore the mode changes sent to it from my radio control when in flight while my onboard computer is connected (via serial - telem2).

This happens both when I am sending commands via the onboard computer / dronekit, and when I am definitely not sending any commands. And no, I am definitely not repeatedly sending mode change requests in my code.

Eventually (after spamming the button) it will change mode, but this is slightly worrying. Has anyone else experienced this?

Thanks

lachlanbrewster1, 2020-12-09:
For anyone in the future coming across this;
I was using a herelink controller, which apparently also uses dronekit/pymavlink/something to talk to the flight controller in some instances. This means that it connects using the same system source as what my code tries to use (the default is 255), so my connection and the herelink connection were arguing/fighting with eachother over who gets to talk to the flight controller. Changed source_system=1 when I connect and it worked out.

closed 2020-12-09 by lachlanbrewster1

kpguru20001, 2021-02-04:
@lachlanbrewster1 can u share the piece of code u used to connect dronekit and herelink

munsterlander, 2021-11-21:
Please with the code. Herelink is causing me fits.