dronekit/dronekit-python #1082: Waiting for mode change forever in vehicle_state example

lukelu0520, 2020-12-18:
Dear dronekit-python friends,

We've follow the example of dronekit-python/examples/vehicle_state/ to connect to a simulator started/managed by the script by
`python vehicle_state.py`

However, we stuck by the message "Waiting for mode change ...".  Is it expected ? Maybe we miss something.

Hope to hear your expertise.

-- Luke

orangefabri, 2022-10-16:
maybe it is same problem with me. pymavlink version problem. did you solved?

Abhayindia, 2026-09-20:
root cause: modern pymavlink's `set_mode_apm` sends `COMMAND_LONG`/`MAV_CMD_DO_SET_MODE`, and the 2015 APM:Copter 3.3 binary that `dronekit-sitl` bundles rejects it - the vehicle replies `COMMAND_ACK(176, 3)` = `MAV_RESULT_UNSUPPORTED`, dronekit never inspects the ack, so the mode silently never changes and the example's untimed `while` loop spins forever (288 iterations over 300s here).
confirmed live in one session: `vehicle.mode = VehicleMode('GUIDED')` left it in STABILIZE, then the legacy `mav.set_mode_send(target, MAV_MODE_FLAG_CUSTOM_MODE_ENABLED, 4)` switched it to GUIDED immediately - so it is the DO_SET_MODE path, not mode numbering (GUIDED is 4 in both old and new mappings). @Al3xFreeman this is why.