dronekit/dronekit-python #1083: Sitl Taking off dronekit python example give arming errors

Nilasstohr, 2020-12-21:
Hi Guys,

I wanted to try the dronekit-sitl before running dronekit on my real quadcopter. So I installed dronekit and drone-sitl on ubuntu 20.04.1 LTS, by following the guide: _https://dronekit-python.readthedocs.io/en/latest/guide/quick_start.html_

Then i did the following:
1. Started dronekit-sitl the from terminal by the command "dronkit-sitl copter"  
2. I then ran Taking off python example from _https://dronekit-python.readthedocs.io/en/latest/guide/taking_off.html_
   using python3

The output of taking off dronekit python example is:
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
Arming motors
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
ERROR:autopilot:ARMING MOTORS
CRITICAL:autopilot:GROUND START
 Waiting for arming...
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
CRITICAL:autopilot:Initialising APM...
Taking off!
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.01
 Altitude:  0.02
 Altitude:  0.02
 Altitude:  0.03
ERROR:autopilot:DISARMING MOTORS
 Altitude:  0.03
 Altitude:  0.01
 Altitude:  0.0
 Altitude:  0.0

From the output it seems like the first 5 lines are stated as critical for some reason dont know if it matters. 
It then manage to pass the pre-arm checks and starting arming unstill the ERROR:autopilot:ARMING MOTORS is returned. It then manage to arm anyway and takeoff until the it disarm after 10 second.
 
I ran the example multiple times and disarming always occurs after 10 seconds. 

what is going on here? does the sitl and taking off example works out of the box or do I need to do something else?
looking forward to hear from you guys.

br.

Nilasstohr, 2020-12-21:
Additional version info
dronekit version: 2.9.2
dronekit sitl version:  copter-3.3

Nilasstohr, 2020-12-21:
just discovered the issue below:
https://github.com/dronekit/dronekit-python/issues/1066
seem like several problem experience, do the developer have a solution?

Nilasstohr, 2020-12-21:
#1066 solutions suggestion did not solve the problem for me.

Nilasstohr, 2020-12-21:
I just tried to run the taking off example on my real quad, it does not there either. @peterbarker can you please provide us with info of what to do? I saw that the last response was back in juli, can we expert a new release soon? Or is there a workaround. It seem like we cant use dronekit for much without the ability to change to guided/auto mode.

Nilasstohr, 2020-12-22:
I think I discovered something interesting, I read from https://ardupilot.org/copter/docs/arming_the_motors.html that arming can only be done in certain modes (guided mode not being one of them). The dronekit taking off sample code actually changes to guided mode and then arm, which may be the reason why there are arming problems running this example. 

So I tried to change the order of the example as seen below, and sitl manage to take-off to the desired high. 
 print "Arming motors"
 vehicle.armed   = True
 # Confirm vehicle armed before attempting to take off
 while not vehicle.armed:
        print " Waiting for arming..."
        time.sleep(1)
 # Copter should arm in GUIDED mode
 vehicle.mode    = VehicleMode("GUIDED")

So does the dronekit taking off example need to be changed, or something else? 
I must say that I am a little disappointed that no one got any clearance on this for almost 6 month.
I wonder if dronekit development is still a live?

Nilasstohr, 2020-12-22:
I think I discovered something interesting, I read from https://ardupilot.org/copter/docs/arming_the_motors.html that arming can only be done in certain modes (guided mode not being one of them). The dronekit taking off sample code actually changes to guided mode and then arm, which may be the reason why there are arming problems running this example. 
So I tried to change the order of the example as seen below, and sitl manage to take-off to the desired high. 
-------------------------------------------
vehicle.armed   = True
while not vehicle.armed:
        print " Waiting for arming..."
        time.sleep(1)
 vehicle.mode    = VehicleMode("GUIDED")
-------------------------------------------
So does the dronekit taking off example need to be changed, or something else? 
I must say that I am a little disappointed that no one got any clearance on this for almost 6 month.
I wonder if dronekit development is still a live?

Nilasstohr, 2020-12-23:
Ok, so i think my problem was that dronekit was not pointing to the correct version of pymavlink after i downgraded to pymavlink 2.4.8 so fix this. Now it seem to takeoff (though with severel errors as seem in log below)

So the fix here was as in #1066 to downgrade from pymavlink 2.4.12 to 2.4.8. (and make sure dronekit is using 2.4.8 not something else)

So
" @peterbarker can you please provide us with info of progress? I saw that the last response was back in july, can we expert a new official update soon?" 

Log output:
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
Arming motors
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
ERROR:autopilot:ARMING MOTORS
CRITICAL:autopilot:GROUND START
 Waiting for arming...
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
CRITICAL:autopilot:Initialising APM...
 Waiting for arming...
Taking off!
VehicleMode:STABILIZE
 Altitude:  0.0
VehicleMode:GUIDED
 Altitude:  0.0
VehicleMode:GUIDED
 Altitude:  0.19
VehicleMode:GUIDED
 Altitude:  1.85
VehicleMode:GUIDED
 Altitude:  3.77
VehicleMode:GUIDED
 Altitude:  5.87
VehicleMode:GUIDED
 Altitude:  8.63
VehicleMode:GUIDED
 Altitude:  10.87
VehicleMode:GUIDED
 Altitude:  13.11
VehicleMode:GUIDED
 Altitude:  15.34
VehicleMode:GUIDED
 Altitude:  17.98
VehicleMode:GUIDED
 Altitude:  19.28
Reached target altitude
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds
ERROR:autopilot:DISARMING MOTORS

danammeansbear, 2021-01-15:
How did you make sure it was pointing to the right version of pymavlink

Nilasstohr, 2021-01-16:
Try the following (importance that you use sudo)

sudo apt-get install python-pip3 python-dev
sudo pip3 install dronekit
sudo pip3 install dronekit-sitl
sudo pip3 uninstall pymavlink          (this removes the new buggy version introduced from 2.4.9)
sudo pip3 install pymavlink==2.4.8
sudo install spyder   (or what you prefer)

from spyder open your script
right click **connect**(...) function and choose "go to defination" it will jump into dronekit library function __init__.py
find and then right click on the function   **MAVConnection**(ip, baud=baud, source_system=source_system,...) and choose "go to definition" it will jump into the pymavlink.py file it uses. Now right on the pymavlink.py folder tab and choose in "show en extrnal file folder", then then go to disp-packages folder (to levels back) here you can locate your mavlink version used by dronekit something like pymavlink.py-2.4.8.dis-info

danammeansbear, 2021-01-16:
Thank you so much!

Nilasstohr, 2021-01-19:
You are welcome, hope it solved your problem :)

o-gent, 2021-02-17:
I've seen this too. I use https://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html this method instead with WSL. You can then use arducopter 4.1 rather than 3.3 which isn't very representative at this point.

IsmailAyvaz, 2021-03-08:
> You are welcome, hope it solved your problem :)

can you make a pull request for your fix ?

danammeansbear, 2021-06-25:
Just want to add, I seeminly only run into this problem when I skip running the simple scripts before branching off other peoples and getting the system confused. Be patient and dont try and do something you arent ready for but also dont stop challenging yourself. its a slow build.

Git-Ilan, 2021-10-03:
I noticed that it is impossible to use pymavlink 2.4.8 with Python 3.7 and above. I had to use Python 3.6 to install it successfully.

prithvi2k2, 2022-12-08:
> I happened to encounter this problem today when I was using pymavlink 2.4.16 along with python 3.9. I did as said and downgraded pymavlink to version 2.4.8, but it did not work with Python 3.7 (and above). I had to use Python 3.6 to install it successfully. I write this down in case another one will run into the same problem as me.

I tested @Git-Ilan 's suggestion and can confirm it works, I downgraded to python 3.6 from 3.9 and also downgraded pymavlink to version 2.4.8 -- It worked; Still hope that dronekit will support python 3 officially anytime later.
Also, I didn't do any rigorous testing, I just tried to takeoff dronekit-sitl's  sim, it worked.

ItayShiff, 2023-09-28:
Any update on this?

danammeansbear, 2023-09-29:
> C:\Users\hp>pip install pymavlink==2.4.8
> Collecting pymavlink==2.4.8
> Using cached pymavlink-2.4.8.tar.gz (3.6 MB)
> Installing build dependencies ... done
> Getting requirements to build wheel ... done
> Installing backend dependencies ... done
> Preparing metadata (pyproject.toml) ... done
> Requirement already satisfied: future in
> c:\users\hp\appdata\local\programs\python\python311\lib\site-packages (from
> pymavlink==2.4.8) (0.18.3)
> Requirement already satisfied: lxml in
> c:\users\hp\appdata\local\programs\python\python311\lib\site-packages (from
> pymavlink==2.4.8) (4.9.3)
> Building wheels for collected packages: pymavlink
> Building wheel for pymavlink (pyproject.toml) ... error
> error: subprocess-exited-with-error
>
> × Building wheel for pymavlink (pyproject.toml) did not run successfully.
> │ exit code: 1
> ╰─> [75 lines of output]
> ###################################
> Skipping mavnative
> ###################################
> running bdist_wheel
> running build
> [... 132 lines cut ...]
> ERROR: Could not build wheels for pymavlink, which is required to install
> pyproject.toml-based projects
>
> when I tried installing pymavlink==2.4.8, this was the error I got, any
> comments or suggestion to resolve this, I am using latest version of python

you will need to create a specific python ENV for your development. This
can be done using anaconda or this can be done using docker. once you have
the correct version of python installed you should be able to follow the
steps previously listed and get working!

*Adam Dabdoub*
Software Developer
c: 513.886.0301

prithvi2k2, 2023-09-29:
@ItayShiff I don't think so it's a problem from dronekit... Please read this for further understanding - https://github.com/dronekit/dronekit-python/issues/484#issuecomment-377809030 

Afaik, it's the pymavlink breaking dronekit when I tested it on windows(don't know about other platforms) back then, i don't exactly know what's the situation now.

What i did to fix was downgrading python version using PyLauncher and also installing lower version of pymavlink as I mentioned in previous comment.

I've also read somewhere that they aren't going to upgrade docs for python3, so maybe they aren't working on this. They still mention python 2.x in docs...  https://dronekit-python.readthedocs.io/en/latest/develop/installation.html

Usbasuthkar, 2023-09-30:
@prithvi2k2 It says that python version of 3.6 and below have come to an end and I am not able to download it, I tried finding other sources but it didn't work, can you please give some reliable source where I can get python 3.6 and below, thanks a lot in advance

Usbasuthkar, 2023-09-30:
@prithvi2k2 As you said, I downloaded the older version of python, it was python 2.7.18 and I got pymavlink of version 2.4.8, thanks a lot for that, now I wrote the command --
```
mavlink.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14550
```
another command which I wrote was - 
```
python mavlink.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14550
```
but it didn't do anything, Is there a different command??

Usbasuthkar, 2023-10-05:
Thanks a lot guys it worked, what I did was just reduced my python version to 2.7.18, pymavlink to 2.4.8, mavproxy to 1.8.22 and for trial I just used a python code from dronekit documentation.

viaSeunghyun, 2023-10-06:
Looking at the release notes, it officially supports python 3.(after 2.9.1)
However, I don't think it's supported for all versions.
This is a problem that many python 2.7 based projects have when extending to python 3.

However, many Linux kernels are also adopting python3.x as the stable version of system python, so dronekit-python will have to change as well.

wangcongai, 2024-05-15:
After pymavlink version==2.4.8, the set_mode_apm function in pymavlik.mavutil.py changed, which cause failure on ardupilot MODE CHANGE.
In order to use new version like pymavlink version==2.4.41, you just need to comment the code on "mav.command_long_send" and change it back to old "mav.set_mode_send". Reboot your IDE, Then the problem solved!
[image]