dronekit/dronekit-python #1084: WARNING:autopilot:Mission upload timeout

iaomari93, 2021-01-04:
I have pixhawk 2.4.8 and I'm trying to upload mission using the upload() function. I'm following the same order in the examples.

running the code on the ArduCopter sim_vehicle.py simulator every thing works fine. but when I try the same code with the pixhawk I got WARNING:autopilot:Mission upload timeout and after few seconds I got /__init__.py", line 3031, in upload
    raise TimeoutError
dronekit.TimeoutError.
the connection with the drone works fine as well as arm and takeoff function.
I tried to change the timeout to None but that didn't help.

does anyone had this before?
thanks

eminakgun, 2021-01-13:
Please try my implementation to fix this issue #1039

wynn4, 2021-06-05:
I also have this problem even after applying the implementation of @eminakgun . I’m running ArduCopter 4.0.3 and connecting via UDP to a companion computer-based mavlink router (similar to mavproxy) which is connected to the Cube autopilot via serial. Similar to @iaomari93 , everything works fine in SITL. My dronekit app is executed in python2.7

brunoolivieri, 2021-06-05:
@wynn4 I also have this issue. I am not using dronekit, but using pure pymavlink. 

In STIL "everything" works (takeoff, land, GUIDED, parameters, etc.).  In a real board/quadcopter only upload mission have problems: https://discuss.ardupilot.org/t/mavlink-mission-upload-timeout-error-in-copter-4-1-0-dev/71296

(I am using a regular RPi)

eminakgun, 2021-06-06:
Been a long time when I encountered the timeout problem but it was happening during mission download. And I don't really remember all the things. I dig through the debug sessions and noticed that once download process is started, autopilot does not respond sometimes and the dronekit sits and wait for it to respond without doing nothing. But MAVLink protocol requires the requesting side must re-send the request when timeout expires. After that I implemented the correct behaviour. [https://mavlink.io/en/services/mission.html#download_mission](url)

I suggest you to go through the debug sessions and realize what is happening or missing actually. There might be a problem in your physical medium or anything else rather than the software. 

Since dronekit is not officially supported/maintained anymore, you have to go and find it, fix it eventually. And please put it back to the github so that others will benefit from it. I don't think dronekit or MAVLink python libraries are that confusing, just give it a time.

Good luck.