dronekit/dronekit-python #1104: drone not moving

Skided, 2021-04-12:
so im using d4rc drone(maby thats the problem) and when i try to turn on program the log is that:
`Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
Arming motors
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
ERROR:autopilot:ARMING MOTORS
CRITICAL:autopilot:GROUND START
 Waiting for arming...
CRITICAL:autopilot:Initialising APM...
 Waiting for arming...
Taking off!
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
ERROR:autopilot:DISARMING MOTORS
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.0
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
 Altitude:  0.01
`
i just want to see the motors moving but they arent someone can help me with that? the code is on the example page

Serzh09tish24, 2021-07-30:
I have the same problem. I was able to solve the problem just by installing mavproxy CORRECTLY!

Serzh09tish24, 2021-07-30:
L had the same problem. I was able to solve the problem just by installing mavproxy CORRECTLY! I recommend  tou to do this command(for ubuntu, linux)       1) sudo apt-get install python3-dev python3-opencv python3-wxgtk4.0 python3-pip python3-matplotlib python3-lxml python3-pygame
2)pip3 install PyYAML mavproxy --user
3)echo "export PATH=$PATH:$HOME/.local/bin" >> ~/.bashrc
if you had another system, so search in google mavproxy(ardupilot)