dronekit/dronekit-python #1112: pyinstaller connection issue....

KimJinHyeok1, 2021-06-23:
hello now I'm develop the GCS program using dronekit and PyQT5, pyinstaller

in the pycharm everything work well!

so i make .exe file using Pyinstaller

but when i click Connect button (vehicle=connect(udpin:localhost:port))

In the console print "warning: unable to load xml validator libraries. xml validation will not be performed"

I'm doing almost solution in the internet ( ex: hidden import lxml, lxml.etree...., reinstall pyinstaller,pymavlink,mavutil...)
but error is same..

I'm really don't know why that error occur only in the .exe file..

Pleeeeeeeeeease. Let me know any solution..... 

Thankyou

blisshuang, 2022-03-15:
i have same problem

KimJinHyeok1, 2022-03-15:
Hi.
I solved that problem.
Try Using CXFreeze tool,
maybe all works Fine.

good luck :)

Reksin-Aniez, 2022-12-28:
> hello now I'm develop the GCS program using dronekit and PyQT5, pyinstaller
>
> in the pycharm everything work well!
>
> so i make .exe file using Pyinstaller
>
> but when i click Connect button [...]

Can you share GCS details with me. I am also working on similar project

gregor-coder, 2026-10-28:
This warning is caused by a missing mavschema.xsd resource file in your PyInstaller distribution. pymavlink uses this XML Schema Definition (XSD) file to validate MAVLink XML dialects.

Add `--collect-data pymavlink` to pyinstaller command line.

This tells PyInstaller to collect package data files from pymavlink, including mavschema.xsd when it is present in the installed package.

fahad85reham-ctrl, 2026-10-29:
> This warning is caused by a missing mavschema.xsd resource file in your PyInstaller distribution. pymavlink uses this XML Schema Definition (XSD) file to validate MAVLink XML dialects.
>
> Add [...]

Fahad1996

fahad85reham-ctrl, 2026-10-29:
html {
  height: 100%;
}

```
body {
  min-height: 100%;
  display: flex;
  justify-content: center; /* Horizontal centering */
  align-items: center;     /* Vertical centering */
  margin: 0;
}
```