dronekit/dronekit-python #1114: WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds - connect() Dronekit does not work

ce8, 2021-07-01:
Hi,
I started to using the Dronekit, Dronekit-STIL and Mavlink to simulate my python scripts. Afters some days using it without problem I started to receive the error:  WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds.

I had tried to reinstall all the things but nothings works.

I had install the PIP pachages on Linux Ubutun 18.  I had try the same packages on Ubutun 20 but I receive the same error.

I had install this packages:
pymavlink>=2.3.3
MAVProxy-1.8.39
dronekit-2.9.2
dronekit-sitl-3.3.0
Python 2.7.17 

Follow my steps to receive the error:

1 - dronekit-sitl copter --home=-25.56731,-42.61554,0,180

os: linux, apm: copter, release: stable
SITL already Downloaded and Extracted.
Ready to boot.
Execute: /home/cesar/.dronekit/sitl/copter-3.3/apm --home=-23.56731,-46.61554,0,180 --model=quad -I 0
Started model quad at -23.56731,-46.61554,0,180 at speed 1.0
bind port 5760 for 0
Starting sketch 'ArduCopter'
Serial port 0 on TCP port 5760
Starting SITL input
Waiting for connection ....
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763

2 - mavproxy.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14551 --out udp:10.0.2.15:14550

Connect tcp:127.0.0.1:5760 source_system=255
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
online system 1
STABILIZE> Mode STABILIZE
AP: Calibrating barometer
AP: Initialising APM...
AP: barometer calibration complete
AP: GROUND START
Init Gyro**
INS
----------------------------------------
G_off: 0.00, 0.00, 0.00
A_off: 0.00, 0.00, 0.00
A_scale: 1.00, 1.00, 1.00

3 -  python hello.py

Start simulator (SITL)
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds

after 30s

ERROR:dronekit.mavlink:Exception in MAVLink input loop
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 211, in mavlink_thread_in
    fn(self)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 1371, in listener
    self._heartbeat_error)
APIException: No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "hello.py", line 11, in <module>
    vehicle = connect(connection_string, wait_ready=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 3166, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2275, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

aysegul-sarikaya, 2021-07-07:
I am getting the same error.
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds.

Did you find a solution to the problem?

arBalasquide, 2021-07-30:
@ce8 Hello!

Try changing the connect string from `tcp:127.0.0.1:5760` to `udp:127.0.0.1:14551` and let me know if that works.

harmor123, 2024-04-13:
I'm also having the same issue, have you solved it？Thank you!
[image]

python3:
```
from dronekit import connect

# Connect to the Vehicle (in this case a UDP endpoint)
vehicle = connect('/dev/ttyUSB0', wait_ready=True, baud=921600)

# vehicle is an instance of the Vehicle class
print("Autopilot Firmware version: %s" % vehicle.version)   #飞控版本
print("Autopilot capabilities (supports ftp): %s" % vehicle.capabilities.ftp) 
print("Global Location: %s" % vehicle.location.global_frame )    #全球定位信息（经纬度和高度）
print("Global Location (relative altitude): %s" % vehicle.location.global_relative_frame )  #相对坐标
print("Local Location: %s" % vehicle.location.local_frame)     #NED相对位置信息，东南西北
print("Attitude: %s" % vehicle.attitude)     #无人机朝向：roll,pitch,yaw单位弧度（-π~π）
print("Velocity: %s" % vehicle.velocity)     #三维速度（m/s）
print("GPS: %s" % vehicle.gps_0)             #GPS信息
print("Groundspeed: %s" % vehicle.groundspeed)          #地速（m/s）
print("Airspeed: %s" % vehicle.airspeed )               #空速（m/s）
print("Gimbal status: %s" % vehicle.gimbal )            #云台信息（单位度）
print("Battery: %s" % vehicle.battery      )            #电池信息
print("EKF OK?: %s" % vehicle.ekf_ok  )                 #扩展卡尔曼滤波器信息
print("Last Heartbeat: %s" % vehicle.last_heartbeat) 
print("Rangefinder: %s" % vehicle.rangefinder  )        #超声波或激光雷达模块状态
print("Rangefinder distance: %s" % vehicle.rangefinder.distance  )    #超声波或激光雷达模块的传感距离
print("Rangefinder voltage: %s" % vehicle.rangefinder.voltage)        #超声波或激光雷达模块电压
print("Heading: %s" % vehicle.heading )                               #无人机朝向（单位度）
print("Is Armable?: %s" % vehicle.is_armable      )                   #是否可以解锁
print("System status: %s" % vehicle.system_status.state)              #系统状态
print("Mode: %s" % vehicle.mode.name    )                             # settable飞行模式
print("Armed: %s" % vehicle.armed     )                               # settable解锁状态
```