dronekit/dronekit-python #1115: MAVLink object has no attribute 'obstacle_distance_encode'

lachlanbrewster, 2021-07-08:
Going off the instructions on the message_factory attribute documentation in dronekit __init__.py, this should work:

vehicle.message_factory.obstacle_distance_encode

as obstacle_distance is listed as a valid message on the mavlink message definitions page.
( https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml )

However it seems to not be implemented. I assume my other options are to use pymavlink to send a message outside of dronekit, however this isn't ideal. Any ideas?

lachlanbrewster, 2021-07-08:
I have found that it is not implemented in the current version of pymavlink by default, you have to force it to use MAVLink 2 as described in the 'Choosing the Dialect/MAVLink Version' section at https://mavlink.io/en/mavgen_python/ . 

However I can't find a way to force dronekit to use MAVLink 2, which seems like a pretty big issue. This does seem to work though: https://github.com/ArduPilot/pymavlink/issues/389 .

```
from pymavlink.dialects.v20 import ardupilotmega as mavlink2

msg = mavlink2.MAVLink_obstacle_distance_message(
                    0, #time
                    3, #sensor type
                    abc, # abc is usually the array of 72 elements
                    1, #angular width
                    10, #min distance
                    maxDist, #max distance
                    1.53,
                    -55.08,
                    12
                    )
  time.sleep(0.01)
  master.mav.send(msg)   # or use vehicle.send_mavlink(msg)
```

lachlanbrewster, 2021-07-09:
I can force dronekit to use Mavlink 2  by setting these environment variables: 

os.environ['MAVLINK20'] = '1'
os.environ['MAVLINK_DIALECT'] = 'ardupilotmega'

Then obstacle_distance_encode works.

This seems to work for now but doesn't seem very reliable.

closed 2021-07-15 by lachlanbrewster

rizvanovic, 2022-06-22:
I assume this never got resolved other than your stated fix?