dronekit/dronekit-python #1118: Issues arming and changing mode using dronekit

SohaibAJ, 2021-07-23:
Hello, I have been trying to run a simple dronekit code but have been running into quite a few problems. Firstly I was unable to change the flight mode using the mavproxy command line and the dronekit code but by adding a snippet in the mavutil.py file I was able to change the mode in the mavproxy.py command line. Running dronekit sitl works and changes the modes but when I try to run the same code on the actual drone it give me the following error in dronekit command windows

WARNING:autopilot:Flight mode change failed

and this error in the mavproxy UI whenever I want to change the flight modes
Got COMMAND_ACK: DO_SET_MODE: UNSUPPORTED

arBalasquide, 2021-07-23:
@SohaibAJ Hey!

Can you share your code?

This could be for a few reasons depending on what you're telling the drone to do.

SohaibAJ, 2021-07-24:
@arBalasquide Sure. The code is just the simple arm and takeoff code

`from dronekit import connect, VehicleMode, LocationGlobalRelative
from pymavlink import mavutil
import time

import argparse
parser = argparse.ArgumentParser()
parser.add_argument('--connect', default='127.0.0.1:5760')
args = parser.parse_args()

# Connect to the Vehicle
print ('Connecting to vehicle on: %s' % args.connect)
vehicle = connect(args.connect, baud=57600, wait_ready=True)

# Function to arm and then takeoff to a user specified altitude
def arm_and_takeoff(aTargetAltitude):

  print ("Basic pre-arm checks")
  # Don't let the user try to arm until autopilot is ready
  #while not vehicle.is_armable:
  #  print (" Waiting for vehicle to initialise...")
  #  time.sleep(1)
  

  vehicle.mode    = VehicleMode("STABILIZE")

  print ("Arming motors")
  vehicle.flush()
  print(vehicle.armed)
  vehicle.armed = True
  print(vehicle.armed)
  
  time.sleep(2)
  # Copter should arm in GUIDED mode
  vehicle.mode    = VehicleMode("GUIDED")
  time.sleep(1)

  while not vehicle.armed:
    print (" Waiting for arming...")
    time.sleep(1)

  print ("Taking off!")
  vehicle.simple_takeoff(aTargetAltitude) # Take off to target altitude

  # Check that vehicle has reached takeoff altitude
  while True:
    print (" Altitude: ", vehicle.location.global_relative_frame.alt )
    #Break and return from function just below target altitude.
    if vehicle.location.global_relative_frame.alt>=aTargetAltitude*0.95: 
      print ("Reached target altitude")
      break
    time.sleep(1)

##----------------Above code was provided by Dronekit itself

#----begin programming form here

##-------------------------------------------------TAKING OFF AT 20M
arm_and_takeoff(10)

##------------------------------------------------hOVERING FOR 10SEC BEFORE WE LAND
time.sleep(5)

##------------------------------------------------LAND

vehicle.mode = VehicleMode("LAND")
time.sleep(1)
while (vehicle.location.global_relative_frame.alt > 0.1):
  print("waiting")
  time.sleep(1)

##----------------------------------------------------Landed
vehicle.close()`

arBalasquide, 2021-07-30:
@SohaibAJ Sorry for late reply.

Why comment this out? If we try to arm before we're ready then arming will fail and we'll have to manually arm the drone.
```
#while not vehicle.is_armable:
#  print (" Waiting for vehicle to initialise...")
#  time.sleep(1)
```

Why are we trying to arm in Stabilize when we want to take off in Guided? Also, the sleep functions do not guarantee we'll be armed by then.
```
vehicle.mode    = VehicleMode("STABILIZE")

print ("Arming motors")
vehicle.flush()
print(vehicle.armed)
vehicle.armed = True
print(vehicle.armed)
  
time.sleep(2)
# Copter should arm in GUIDED mode
vehicle.mode    = VehicleMode("GUIDED")
time.sleep(1)
```

Perhaps we should change to
```
vehicle.mode    = VehicleMode("GUIDED")
vehicle.armed   = True

# Confirm vehicle armed before attempting to take off
while not vehicle.armed:
    print("Waiting for arming...")
    time.sleep(1)

print("Taking off!")
vehicle.simple_takeoff(aTargetAltitude)
```
as provided in https://dronekit-python.readthedocs.io/en/latest/guide/taking_off.html

epkeefe123, 2024-06-26:
> @SohaibAJ Sorry for late reply.
>
> Why comment this out? If we try to arm before we're ready then arming will fail and we'll have to manually arm the drone.
>
> ```
> #while not vehicle.is_armable:
> # [...]

can you help me in my script i cant get it to change the mode and set it to land at the end of the script

#1222

hamishwillee, 2024-06-26:
I'm closing this. The original post was a not supported response from a command. That means the command is not supported. I suspect that whatever you are talking to supports the SET_MODE message rather than the DO_SET_MODE command mode used by DroneKit. If it is ardupilot, then I expect it would work on a recent version as they moved to the command over the last few years.

closed 2024-06-26 by hamishwillee