dronekit/dronekit-python #1119: Can I do a circle on guided mode?

YusufMuharremDaskaya, 2021-08-03:
I am using an SITL and dronekit and I want to do a circle. I tried MAV_CMD_NAV_LOITER_TIME command but with this paramaters : 10, 0, 0, 0, -35.1243080, 149.90123890, 10 
But nothing happend. Am I doing something wrong? or is there an other easier way?

arBalasquide, 2021-08-04:
@YusufMuharremDaskaya Hey!

https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TIME

What vehicle are you using? Message states "Multicopter vehicles stop at the point (within a vehicle-specific acceptance radius)." Thus, I don't think copters go around in circles in loiter mode with this command.

If you're using a "forward moving vehicle", then you can continue.

Param 1: Time to loiter
Param 3: Radius in meters of loiter circle

You set parameter 3 to 0 and thus a circle of 0 radius.

Try changing it to something else and see if it works.

Otherwise, if using a copter, I created a small script that'll make the drone go around in a circle. There seems to be a Circle mode for copters but when I tested it, it kept crashing in the SITL.

YusufMuharremDaskaya, 2021-08-04:
@arBalasquide I am using a copter, and in my SITL I can't use Circle mode too. Did you shared your script? I can't find it in your repos.

arBalasquide, 2021-08-04:
@YusufMuharremDaskaya 

Wrote it quickly so theres definitely a lot of improvements to be done. But shows that it works in the SITL so good enough for now. 
```python3 
#!/usr/bin/env python3

import math
from time import sleep
from dronekit import connect, VehicleMode, LocationGlobalRelative, Command
from pymavlink import mavutil

if __name__ == "__main__":
    vehicle = connect("udp:127.0.0.1:14550", wait_ready=True)

    while not vehicle.is_armable:
        sleep(1)

    vehicle.mode = VehicleMode('GUIDED')
    while not vehicle.armed:
        vehicle.armed = True
        sleep(1)

    alt = 10
    radius = 0.0001

    lat = vehicle.location.global_relative_frame.lat
    lon = vehicle.location.global_relative_frame.lon

    vehicle.simple_takeoff(alt)

    while True:
        if vehicle.location.global_relative_frame.alt >= alt * 0.95:
            break
        sleep(1)

    points = 30
    loops = 2 * points # do two loop
    coords = []
    for i in range(0, loops):
        degrees = (i/points)*360
        radians = (math.pi/180)*degrees
        x = lat + radius * math.cos(radians)
        y = (lon + radius * math.sin(radians))
        coords.append((x,y))

    cmds = vehicle.commands
    cmds.clear()

    for lat,lon in coords:
        point = LocationGlobalRelative(lat,lon,alt)
        cmds.add(Command( 0, 0, 0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, mavutil.mavlink.MAV_CMD_NAV_WAYPOINT, 0, 0, 0, 0, 0, 0, lat, lon, alt))

    cmds.upload()

    vehicle.mode = VehicleMode("AUTO")
    sleep(1)

    while True:
        nextwaypoint=vehicle.commands.next
        sleep(1)
```

prithvi2k2, 2023-01-21:
@arBalasquide Can you explain how does `radius = 0.0001` in your custom script for copters work? What units does it consider... is it in meters?? It's not clear for me, I would like to set custom radius either in centimeters or meters somehow... how would I do that?