dronekit/dronekit-python #1126: Is DroneKit still actively being developed?

RickerStem, 2021-10-15:
Hi DroneKit users,

At first, great piece of software  (y)

But I was wondering what is the status/roadmap of this project. Is it still wise to use this or are there any other better alternatives?

I was wondering this because:
- The forum isn't working 
- Can't find many recent Youtube videos about it? 
- No Python 3 support?
- Latest release 2017

DroneDojo Youtube Channel has some great videos on DroneKit
But overall it just doesn't seem so popular?

Maybe somebody can explain or give advice on alternatives?

Thank you in advance.

peterbarker, 2021-10-15:
On Fri, 15 Oct 2021, RickerStem wrote:

> But I was wondering what is the status/roadmap of this project. Is it still wise to use this or are there any other better alternatives?

Essentially unmaintained.  I don't have any experience with similar 
frameworks apart from raw pymavlink to offer any advice on alternatives.

>  *  No Python 3 support?

It should work just fine with Python3.  If it doesn't somewhere I'll patch 
it for that.

hamishwillee, 2021-10-16:
[MAVSDK](https://mavsdk.mavlink.io/main/en/index.html) is written/tested against PX4 but recently took some code that improves compatibility with ArduPilot. I have done no testing on this but I know the people who integrated that support use both flight stacks, so it will work at least to some level.

As Peter says, this project is abandonware. It does have quite an active community of users on the gitter channel though, and it is not hard to extend to support specific messages etc.

Pascal91, 2021-10-16:
Hello,
Yes I confirm that **DroneKit** works well with **PX4** and **Python 3** (at least for the functions I use).
But, Python moved recently from version **3.9.7** to version **3.10.0**, and it seems that with this latest version something is broken/missing : With **v3.9.7** everything was Ok, but with **v3.10.0** It's no more possible to install the "`dronekit`" (or the "`dronekit-sitl`") module ; the "`pip install dronekit`" command returns an error saying that Microsoft Visual C++ must be installed to compile this or that...
It is possible to update **DroneKit** (and **DoneKit-SITL**) to solve this issue (with Python **3.10.0**) ?
Thanks.

Pascal91, 2021-10-16:
PS: Here is the output (error) of the "`pip install dronekit`" command:

Collecting dronekit
  Using cached dronekit-2.9.2-py3-none-any.whl (56 kB)
Collecting monotonic>=1.3
  Using cached monotonic-1.6-py2.py3-none-any.whl (8.2 kB)
Collecting pymavlink>=2.2.20
  Using cached pymavlink-2.4.17-py3-none-any.whl
Collecting future
  Using cached future-0.18.2-py3-none-any.whl
Collecting lxml
  Using cached lxml-4.6.3.tar.gz (3.2 MB)
Building wheels for collected packages: lxml
  Building wheel for lxml (setup.py): started
  Building wheel for lxml (setup.py): finished with status 'error'
  Running setup.py clean for lxml
Failed to build lxml
Installing collected packages: lxml, future, pymavlink, monotonic, dronekit
    Running setup.py install for lxml: started
    Running setup.py install for lxml: finished with status 'error'

  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Pascal\Documents\Python\Test\venv\Scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Pascal\\AppData\\Local\\Temp\\pip-install-2ermlsa2\\lxml_a7e07c6cb6144923b4f091088a18bf00\\setup.py'"'"'; __file__='"'"'C:\\Users\\Pascal\\AppData\\Local\\Temp\\pip-install-2ermlsa2\\lxml_a7e07c6cb6144923b4f091088a18bf00\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Pascal\AppData\Local\Temp\pip-wheel-wy4kvn43'
       cwd: C:\Users\Pascal\AppData\Local\Temp\pip-install-2ermlsa2\lxml_a7e07c6cb6144923b4f091088a18bf00\
  Complete output (76 lines):
  Building lxml version 4.6.3.
  Building without Cython.
  Building against pre-built libxml2 andl libxslt libraries
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.10
  creating build\lib.win-amd64-3.10\lxml
  copying src\lxml\builder.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\cssselect.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\doctestcompare.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\ElementInclude.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\pyclasslookup.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\sax.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\usedoctest.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\_elementpath.py -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\__init__.py -> build\lib.win-amd64-3.10\lxml
  creating build\lib.win-amd64-3.10\lxml\includes
  copying src\lxml\includes\__init__.py -> build\lib.win-amd64-3.10\lxml\includes
  creating build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\builder.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\clean.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\defs.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\diff.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\ElementSoup.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\formfill.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\html5parser.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\soupparser.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\usedoctest.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\_diffcommand.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\_html5builder.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\_setmixin.py -> build\lib.win-amd64-3.10\lxml\html
  copying src\lxml\html\__init__.py -> build\lib.win-amd64-3.10\lxml\html
  creating build\lib.win-amd64-3.10\lxml\isoschematron
  copying src\lxml\isoschematron\__init__.py -> build\lib.win-amd64-3.10\lxml\isoschematron
  copying src\lxml\etree.h -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\etree_api.h -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\lxml.etree.h -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\lxml.etree_api.h -> build\lib.win-amd64-3.10\lxml
  copying src\lxml\includes\c14n.pxd -> build\lib.win-amd64-3.10\lxml\includes
  copying src\lxml\includes\config.pxd -> build\lib.win-amd64-3.10\lxml\includes
  copying src\lxml\includes\dtdvalid.pxd -> build\lib.win-amd64-3.10\lxml\includes
  copying src\lxml\includes\etreepublic.pxd -> build\lib.win-amd64-3.10\lxml\includes
  copying src\lxml\includes\htmlparser.pxd -> build\lib.win-amd64-3.10\lxml\includes
[... 90 lines cut ...]
    copying src\lxml\includes\xpath.pxd -> build\lib.win-amd64-3.10\lxml\includes
    copying src\lxml\includes\xslt.pxd -> build\lib.win-amd64-3.10\lxml\includes
    copying src\lxml\includes\__init__.pxd -> build\lib.win-amd64-3.10\lxml\includes
    copying src\lxml\includes\etree_defs.h -> build\lib.win-amd64-3.10\lxml\includes
    copying src\lxml\includes\lxml-version.h -> build\lib.win-amd64-3.10\lxml\includes
    creating build\lib.win-amd64-3.10\lxml\isoschematron\resources
    creating build\lib.win-amd64-3.10\lxml\isoschematron\resources\rng
    copying src\lxml\isoschematron\resources\rng\iso-schematron.rng -> build\lib.win-amd64-3.10\lxml\isoschematron\resources\rng
    creating build\lib.win-amd64-3.10\lxml\isoschematron\resources\xsl
    copying src\lxml\isoschematron\resources\xsl\RNG2Schtrn.xsl -> build\lib.win-amd64-3.10\lxml\isoschematron\resources\xsl
[... cut ...] en'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Pascal\AppData\Local\Temp\pip-record-mjwpjsl5\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Pascal\Documents\Python\Test\venv\include\site\python3.10\lxml' Check the logs for full command output.

peterbarker, 2021-10-16:
On Sat, 16 Oct 2021, Pascal91 wrote:

> RUNNING BUILD_EXT
> BUILDING 'LXML.ETREE' EXTENSION
> ERROR: MICROSOFT VISUAL C++ 14.0 OR GREATER IS REQUIRED. GET IT WITH "MICROSOFT C++ BUILD TOOLS": HTTPS://VISUALSTUDIO.MICROSOFT.COM/VISUAL-CPP-BUILD-TOOLS/
> 
> ERROR: Failed building wheel for lxml
> ERROR: Command errored out with exit status 1:

This looks like a problem with a package we depend on, rather than 
dronekit-python itself.  You should be able to get the same failure with 
`pip install lxml`

We don't pin that package, so it should be the most recent being attempted 
here.

hamishwillee, 2021-10-18:
Yes. Pymavlink is the MAVLink library for python generated using mavgen. It has a few helper scripts that implement a small subset of the services provided by MAVLink - to allow working with commands, missions etc., but by no means exhaustive. If it already does what you need though, it is perfectly good and very well supported.

peterbarker, 2021-10-19:
On Tue, 19 Oct 2021, Pascal91 wrote:

> For your information: I've tried to contact the Dev Team of LXML, but their
> email address ***@***.***") is not active anymore.
> If someone can contact them... maybe on their Bug Tracker
> ("https://launchpad.net/lxml")...
> Because as long as they've not solved their "installation/compilation
> issue", DroneKit and DroneKit-SITL won't run with Python 3.10.0.

Thanks for chasing this.

It's actually a rather worrying problem on other fronts - pymavlink also 
uses and rquires lxml >= 3.6.0