dronekit/dronekit-python #1132: AttributeError: module 'collection' has no attribute 'MutableMapping'

Pascal91, 2021-11-06:
Hello,

When importing the **DroneKit** module ("`import dronekit`") under Python **3.10**, the "`AttributeError: module 'collection' has no attribute 'MutableMapping'`" error is raised at the line 2689 ("`class Parameters(collections.MutableMapping, HasObservers):`") of the "`dronekit/__init__.py`" file.
This makes **DroneKit** incompatible (i.e. unusable) with Python **3.10**.

Thanks,
Best regards.

jeisson333, 2021-11-12:
Hello.

x2

Pascal91, 2021-11-17:
If I have correctly understood the error, and the evolution between Python 3.9 and 3.10, then I propose the following correction (to be confirmed by the owners of the module):
Replace "`collections.MutableMapping`" by "`collections.abc.MutableMapping`" (in the "`dronekit/__init__.py`" file).

Pascal91, 2023-03-10:
Hello,

Do you have any idea when a new version (taking into account the "blocking" issues with Python 3.10 _[and now certainly Python 3.11 too]_) will be released, please ?

Thanks.

hamishwillee, 2023-08-03:
@morzack Have you got as far as a "plan" for supporting Python 3 versions? I.e. any kind of ETA?

benner, 2023-10-07:
same question

hamishwillee, 2023-10-08:
@benner Never? Until someone does the work, and there is no team here, it isn't going to happen. @morzack is interested in doing this work but has not yet committed time to it.

morzack, 2023-10-28:
From what I've seen, the most up to date commits on github for dronekit-python work well up to at least Python3.10 -- I can't say that I've exhaustively tested every feature, but the core functionality works. Notably this is not the version that is on pypi, but that's not a big problem.

The main problems that I've seen in experimenting with getting everything up to date + onboarding some new people with dronekit recently lies in the documentation (all of the examples are python2 specific) and in dronekit-sitl being out of date. The only other noticeable problem lies in the CI/unit testing for dronekit-python, which I think boils down to dronekit-sitl not working well in python3 again.

To answer the original question in this thread: pointing the pip install at this GitHub repo should at least give a dronekit version that works for the more recent versions of python3.

@hamishwillee Sorry for the delay here -- I haven't been able to allocate much time to take an in-depth look at what needs to be brought up to date. My best estimate of what needs to be done is above. I.. don't know when I'll have time to actually start running through that list (maybe december?), so again, sorry. I probably continued to stretch myself a bit too thin when mentioning interest in maintaining this 😅

hamishwillee, 2023-11-15:
Thanks @morzack - well, let's continue doing what you can do. 

> To answer the original question in this thread: pointing the pip install at this GitHub repo should at least give a dronekit version that works for the more recent versions of python3.

What do you mean - update the docs? Or just make https://pypi.org/project/dronekit/ have a link to this repo? If we do that, perhaps just do another pypi release with this tree?

Pascal91, 2023-11-15:
Moving a "version" from this **GitHub** Repository to the official **PyPI** Repository would require to Tag and Release, here, a "stable version" (for example "v2.9.3").
So this would require to check (and maybe update) everything (such as, for example, the "tests", the "documentation" and the "examples"), beforehand, which is certainly a significant work.

It's a pitty that **DroneKit** not be "living" and lack Maintainers, because it's certainly the easiest Python Package to communicate via MAVLink.

ZivotJeKrasny, 2024-06-23:
Small trick, but it works (Python 3.12):

```python
import collections

collections.MutableMapping = collections.abc.MutableMapping
from dronekit import connect, VehicleMode

vehicle = connect("/dev/ttyACM0", wait_ready=True)
print("Autopilot Firmware version: %s" % vehicle.version)
```
> Autopilot Firmware version: APM:Copter-4.5.3

ForrestFire0, 2024-07-02:
> Small trick, but it works (Python 3.12):
>
> ```python
> import collections
>
> collections.MutableMapping = collections.abc.MutableMapping
> from dronekit import connect, VehicleMode
>
> vehicle = [...]

Can confirm the import succeeds with this workaround with python 3.11.

DipFlip, 2025-02-13:
For Python 3.10 I had to run the trick slightly differently
```python
from collections import abc
import collections
collections.MutableMapping = abc.MutableMapping
```