dronekit/dronekit-python #1133: how to set velocities of the copter without setting Yaw?

shubham-shahh, 2021-11-09:
Hi, I am using
`
    def set_velocity_body(vehicle, vx, vy, vz):

    msg = vehicle.message_factory.set_position_target_local_ned_encode(
            0,
            0, 0,
            mavutil.mavlink.MAV_FRAME_BODY_NED,
            0b0000111111000111, #-- BITMASK -> Consider only the velocities
            0, 0, 0,        #-- POSITION
            vx, vy, vz,     #-- VELOCITY
            0, 0, 0,        #-- ACCELERATIONS
            0, 0)
    vehicle.send_mavlink(msg)
    vehicle.flush()`

when I give velocity in Y, the copter heads in the direction and changes the yaw to point towards the direction of the velocity given. I don't want that yaw correction

I know It is a abandoned project but any help is appreciated
@peterbarker 

thanks

spacemonk-ey, 2021-12-17:
Hey,
it appears to me, that you have to many bits in your bitmask, haven't you?
Not sure, if this might cause an issue. Or do you have a reason for using 16 bits instead of 12?
You might wanna try 0b110111000111 as bitmask.

shubham-shahh, 2021-12-18:
> Hey, it appears to me, that you have to many bits in your bitmask, haven't you? Not sure, if this might cause an issue. Or do you have a reason for using 16 bits instead of 12? You might wanna try 0b110111000111 as bitmask.

16 bit, bit mask is suggested in dronekit docs

closed 2021-12-18 by shubham-shahh