dronekit/dronekit-python #1134: Compatibility with Python 3.10

Pascal91, 2021-11-14:
Hello,

Is it planned in short (or medium) term, to have **DroneKit** updated to be "compatible" with Python **3.10**?
(Currently, it does not work anymore. It was still Ok with Python **3.9**, but fails with **3.10**, simply right from the "`import dronekit`" instruction.)

_Example of issue to be solved: [AttributeError: module 'collection' has no attribute 'MutableMapping' #1132](https://github.com/dronekit/dronekit-python/issues/1132#issue-1046536695)._

Thanks,
Best regards,
Pascal.

Pushpendra766, 2021-12-04:
I am facing this issue of NoModuleFound: dronekit. What should I do ? Should I downgrade python to 3.9

peterbarker, 2021-12-16:
> _Example of issue to be solved: [AttributeError: module 'collection' has no attribute 'MutableMapping' #1132](https://github.com/dronekit/dronekit-python/issues/1132#issue-1046536695)._

I've committed a patch to master to fix this problem.

I'm contemplating doing another release with the fix included.

Pascal91, 2022-04-16:
Hello Peter,

Do you have any idea when a new version (taking into account the "blocking" issues with Python 3.10) will be released, please ?

Thanks.

StripedMonkey, 2022-09-23:
I'm also interested in seeing a new release, the last official release for dronekit is borderline ancient, and it's not like there has not been any work on it since.

Ubuntu 22.04 specifically is defaulting to python 3.10 making this a more apparent issue.