dronekit/dronekit-python #1138: Not able to set guided mode

kumar-rajwnai, 2021-12-02:
I am trying to run the following script on my laptop.
os : windows
python : 3.6.8

```
#############DEPENDENCIES#######################

from dronekit import connect, VehicleMode,LocationGlobalRelative,APIException
import time
import socket

import math
import pymavlink.mavutil as mavutil
##############FUNCTIONS##########################

##Function to arm the drone props and takeoff at targetHeight (m)

def arm_and_takeoff(targetHeight):

	while vehicle.is_armable!=True:
		print("Waiting for vehicle to become armable.")
		time.sleep(1)
	print("Vehicle is now armable")

	vehicle.mode = VehicleMode("GUIDED")

	while vehicle.mode!='GUIDED':
		print("Waiting for drone to enter GUIDED flight mode")
		time.sleep(1)
	print("Vehicle now in GUIDED MODE. Have fun!!")

	vehicle.armed = True
	while vehicle.armed==False:
		print("Waiting for vehicle to become armed.")
		time.sleep(1)
	print("Look out! Virtual props are spinning!!")

	vehicle.simple_takeoff(targetHeight)

	while True:
		print("Current Altitude: %d"%vehicle.location.global_relative_frame.alt)
		if vehicle.location.global_relative_frame.alt>=.95*targetHeight:
			break
		time.sleep(1)
	print("Target altitude reached!!")

	return None

############MAIN EXECUTABLE#############

####sim_vehicle.py opens up port on localhost:14550

vehicle = connect('127.0.0.1:14550',wait_ready=True)

####Arm the drone and takeoff into the air at 5 meters

arm_and_takeoff(5)
print("Vehicle reached target altitude")

####Once drone reaches target altitude, change mode to LAND 

vehicle.mode=VehicleMode('LAND')
while vehicle.mode!='LAND':
	print("Waiting for drone to enter LAND mode")
	time.sleep(1)
print("Vehicle now in LAND mode. Will touch ground shortly.")
```

log from mavproxy
Got COMMAND_ACK: REQUEST_AUTOPILOT_CAPABILITIES: ACCEPTED
Got COMMAND_ACK: DO_SET_MODE: UNSUPPORTED

log from script
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
Vehicle is now armable
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode
Waiting for drone to enter GUIDED flight mode

can you please help me how can i run the basic program?

Pascal91, 2021-12-06:
Hello,

You have to use "`vehicle.mode.name`"  (to compare with "`GUIDED`") as explained in the documentation and shown in the "[vehicle_state.py](https://github.com/dronekit/dronekit-python/blob/master/examples/vehicle_state/vehicle_state.py)" example.

Best regards,
Pascal.

kumar-rajwnai, 2021-12-07:
Thanks. it's working after downgrade pymavlink.

Pascal91, 2021-12-07:
Why did you have to downgrade **PyMAVLink** ?
Which version do work correctly ? And which version does not ? (Why ?)

kumar-rajwnai, 2021-12-07:
The latest version of PyMAVLink is not working. I am currently using 2.4.8 version. I found this version from one of the issues i can't remember from which issue i found.

spacemonk-ey, 2021-12-16:
I had the same issue and can confirm that downgrading pymavlink was the solution. Not sure though why exactly.

Lixiangminbuaa, 2022-01-22:
> The latest version of PyMAVLink is not working. I am currently using 2.4.8 version. I found this version from one of the issues i can't remember from which issue i found.

It is useful.Thank you very much