dronekit/dronekit-python #1143: Message stream rates are consistently 4x slower than configured, for the first couple of minutes

eblanshey, 2022-02-08:
I'm encountering a strange issue with message transmission rates that I'm not sure is happening due to dronekit or pymavlink.

I connect to my vehicle with `rate=None` so as NOT to send a `request_data_stream_send` message to set the rate of transmission. As per the Ardupilot docs, I have the `SR1_*` parameters set to the transmission rates I want, in order to optimize performance on a group-message level. I do not have any code that sends messages to my controller after connection.

For the first couple of minutes after connecting, the message transmission rates are EXACTLY 4x less than requested. For example: if I set `SR1_EXTRA2` to 1hz, which requests the `VFR_HUD` message one time a second, I will get the VFR message once every 4 seconds, to the T. If I set it to 40hz, I will get 10 messages a second. I confirmed the `SR1_EXTRA2` parameter is not altered during this time. After a couple of minutes, the desired transmission rate kicks in on its own.

Once the transmission rate is correct, I can restart my python application, and transmission rate is reverted to 4x slower again, without restarting my controller. This makes me believe the issue is not with ardupilot running on my controller, but rather somewhere in the library. I tried looking for a mavlink message request from the library that could cause this, to no avail.

If this is a pymavlink issue, let me know and I'll open an issue on their github. Any help would be appreciated!

peterbarker, 2022-02-08:
On Tue, 8 Feb 2022, Ephi Blanshey wrote:

> I'm encountering a strange issue with message transmission rates that I'm not sure is happening due to dronekit or pymavlink.
>
> I connect to my vehicle with rate=None so as NOT to send a [...]

At a guess, something is downloading parameters, mission items or sending 
mavlink ftp replies; while that's going on we reduce streamrates by a 
factor of 4 (cumulatively...)

eblanshey, 2022-02-09:
> At a guess, something is downloading parameters, mission items or sending
mavlink ftp replies; while that's going on we reduce streamrates by a
factor of 4 (cumulatively...)

Oh, I see. I did not see this behavior documented. Would you be able to point me to where in the code it does this lowering of the stream rate, so I can find a workaround or patch it for myself?

Do you have any suggestions for what can be done to prevent this, or to speed up this initial processing? I turned off a lot of MAVLink messages I don't need so there should be no bandwidth issues going at full speed from the start.

peterbarker, 2022-02-10:
On Wed, 9 Feb 2022, Ephi Blanshey wrote:

>       At a guess, something is downloading parameters, mission items or sending
>       mavlink ftp replies; while that's going on we reduce streamrates by a
>       factor of 4 (cumulatively...)
> 
> Oh, I see. I did not see this behavior documented. Would you be able to

Where would you expect to see it documented?  We're always interested in 
PRs to improve our documentation!

> point me to where in the code it does this lowering of the stream rate, 
> so I can find a workaround or patch it for myself?

https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L954

> Do you have any suggestions for what can be done to prevent this, or to 
> speed up this initial processing? I turned off a lot of MAVLink messages 
> I don't need so there should be no bandwidth issues going at full speed 
> from the start.

Well, it really shouldn't take minutes to download parameters and mission 
items!  You probably want to verify that my guess there is true before 
diving into modifying code :-)

If that does turn out to be the problem, you could consider using mavlink 
ftp to transfer mission and parameters - blindingly fast compared to the 
older protocols.

eblanshey, 2022-02-13:
Aha! Thank you for the clarification! It turns out the problem was of my own doing -- it took a few minutes to sync the parameters because I set the SR1_PARAMS option 1hz, so it was downloading one param a second. I erroneously thought all params are sent in a single message (which would be huge, not sure why I thought that) and didn't think that was necessary 10 times a second.

When setting it back to 10hz (the default), normal speed is returned in 18 seconds for me. If I change it to 40hz, it's 5 seconds. This is plenty good enough for me, so no more changes needed.

> Where would you expect to see it documented?  We're always interested in PRs to improve our documentation!

Seeing as it's unrelated to Dronekit I'd say it belongs in the ardupilot or mavlink docs. The mavlink messages and ardupilot params docs are autogenerated I believe so I'm not sure how to edit them, but I'd change the following:

- [REQUEST_DATA_STREAM](https://mavlink.io/en/messages/common.html#REQUEST_DATA_STREAM): The requested message rate description can be updated to say "The requested message rate. Note that the rate may be lower than requested while parameters or waypoints are being sent.)"
- [MESSAGE_INTERVAL](https://mavlink.io/en/messages/common.html#MESSAGE_INTERVAL): Description should say "The interval between two messages. A value of -1 indicates this stream is disabled, 0 indicates it is not available, > 0 indicates the interval at which it is sent. Note that the interval may be lower than requested while parameters or waypoints are being sent."
- [SRx_PARAMS](https://ardupilot.org/plane/docs/parameters.html#sr1-params-parameter-stream-rate-to-ground-station): "Individual parameter stream rate to ground station".
- I created a [small PR](https://github.com/ArduPilot/ardupilot_wiki/pull/4125) in the [Requesting Data](https://ardupilot.org/dev/docs/mavlink-requesting-data.html) page.

Thank you for your help!

closed 2022-02-13 by eblanshey