dronekit/dronekit-python #1148: connect() never returns when vehicle doesn't send parameters

markus-k, 2022-03-11:
I'm trying to receive data from an INAV FlightController with DroneKit using a Raspberry Pi as a companion computer with DroneKit.

When calling `connect()`, it doesn't return though, even though messages are send from the FC. Some debugging revealed that it is waiting on [Vehicle.initialize](https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L2304), more specifically [the last while-loop](https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L2339) waiting for parameters. Removing this loop makes `connect()` return and I can read position data using `vehicle.location.global_frame`.

From what I understand, INAV doesn't send parameter data over MAVLink.

Not sure if waiting for parameters is really required at that point, so maybe it could be removed? Or add a parameter to stop it from waiting on parameters?

Example code:
```python
from dronekit import connect

MAVLINK_DEVICE = "/dev/serial0"
MAVLINK_BAUD = 115200

conn = connect(MAVLINK_DEVICE, baud=MAVLINK_BAUD, wait_ready=[])
# never returns, but shown debug messages like WARNING:autopilot:UNABLE TO ARM
```

EugeneFolkast, 2024-05-22:
Have the same issue, didn't find any solution?

markus-k, 2024-06-04:
Nope, pretty sure I gave up at that point with whatever I was trying to do back then.