dronekit/dronekit-python #1157: Not getting connected to Jetson nano

kkjdl10, 2022-06-20:
I am trying to establish connection between Pixhawk and Jetson Nano using Telem2 port and dronekit.  The baud rate for telem2 port is 921600. I have used the basic dronekit python script  available from the website

from dronekit import connect
vehicle = connect('/dev/ttyUSB0',baud=921600, wait_ready=True)
print("Mode: %s" % vehicle.mode.name)

I am not able to  establish a connection. It is saying that connection timed out. Heart beat not felt. Can you tell how to proceed about. I am new to this. I am not sure how to rectify this issue

naveen-nayan, 2022-07-28:
@kkjdl10 

- Please check which **/dev/tty** port your Pixhawk is connected to Jetson Nano using Telem2
- If you are connecting to GPIO of Jetson Nano then it usually connects to **/dev/ttyTHS1**
- So need to change **/dev/ttyUSB0** to **/dev/ttyTHS1**

kkjdl10, 2022-08-16:
Yes it is connected to telem2 port.  We are making use of ftti cable to do the connections between telem2 and usb port of jetson nano