dronekit/dronekit-python #1163: the meaning of dronekit battery..?

orangefabri, 2022-09-17:
I searched on the web and dig into websites but there is no information of this.

so I create the issue for future student or adventurer.

what is the meaing of vehicle.battery..?

I can print like this :

[image]

but I couldnt get this meaning

because Im using dual lipo battery.

is it average value of all batteries?

orangefabri, 2022-09-19:
it seems that battery voltage indicator is same with average of two battery, and on mission planner.

it seems that there is no problem.

SarinleFreeman, 2022-11-14:
I'm assuming you connect your battery to the power distribution board(PDB) which then connects to your propellers via speed controllers. Your companion computer(rasperry pi etc) then directly/indirectly connects to your PDB.   

Thus the current, voltage readings should be measuring the total voltage and current of system as perThevenin's/Nortons theorem. 

However it does depend on your actual system.

orangefabri, 2022-11-14:
the drone is connected to laptop with telemetry. Im using dronekit for mini GCS.

my drone have two independent battery and independent power module