dronekit/dronekit-python #1169: dronekit-sitl does not change the mode.

orangefabri, 2022-10-16:
Hello.

below code does not change the mode.

def change_mode(self, mode):
        print("Changing to mode: {0}".format(mode))

        self.vehicle.mode = VehicleMode(mode)
        while self.vehicle.mode.name != mode:
            print('  ... polled mode: {0}'.format(mode))
            # self.vehicle.mode = VehicleMode(mode)
            time.sleep(1)

if I run this code, there is only endless print statement "... polled mode : GUIDED"

I found this solution and tried :

https://github.com/dronekit/dronekit-python/issues/1041

but there is no more pymavlink 2.4.8

pymavlink is 2.4.36 already. 2.4.8 is too old and cannot be installed.

orangefabri, 2022-10-16:
I tried so many versions of pymavlink but no works.

pymavlink 2.4.36 ~26 : cannot change mode
pymavlink 2.4.19 <- cannot change mode
pymavlink 2.4.11 <- time out in connection
pymavlink 2.4.0~9 <- pip setup error

orangefabri, 2022-10-16:
https://github.com/dronekit/dronekit-python/issues/1082

this link also have same problem with me.

orangefabri, 2022-10-18:
https://github.com/dronekit/dronekit-python/issues/1138

this link also have same problem with me

SarinleFreeman, 2022-11-14:
Having the same problem

SarinleFreeman, 2022-11-14:
One fix is utilizing Ubuntu on your windows system. After setting up Ubuntu i downgraded pymavlink to 2.4.8 and mav_proxy to 1.7.1

sudo pip uninstall mavproxy
sudo pip install pymavlink==2.4.8
sudo pip install mavproxy==1.7.1

I will continue my analysis and see if i can find a solution for windows.

orangefabri, 2022-11-14:
oh that solution seems to be really good to me

I will try it later.

I will close this issue. thanks

closed 2022-11-14 by orangefabri

Xarbenence, 2023-01-21:
As someone who is also experiencing the same issue:

> 
> pymavlink 2.4.36 - 26 : cannot change mode
> pymavlink 2.4.19 : cannot change mode
> pymavlink 2.4.11 : time out in connection
> pymavlink 2.4.0 - 9 : pip setup error

but on a Mac with M1, how would you advise me to work around this? I've tried to get Ubuntu up and running via virtual box, but even that isn't seeming to work. Any other recommendations and or versions of Ubuntu that are known to work well on Mac?