dronekit/dronekit-python #1170: Can't take off using dronekit-python

mana747, 2022-10-18:
Drone in use: X500 V2
Flight controller: Pixhawk6c
Firmware: PX4

I'm trying to take off a drone using dronekit-python, but it says "Waiting for arming" and the motors won't move. Is there any solution? The executed program and the log after execution are shown below.

takeoff.py
#!usr/bin/env python
# -*- coding: utf-8 -*-
print( "dronekitスタート" )    # 開始メッセージ
 
from signal import signal, SIGINT   # Ctrl+C(SIGINT)の送出のために必要
from dronekit import connect        # connectを使いたいのでインポート
from dronekit import VehicleMode    # VehicleModeも使いたいのでインポート
from dronekit import LocationGlobal, LocationGlobalRelative   # ウェイポイント移動に使いたいのでインポート
import time                         # ウェイト関数time.sleepを使うために必要
 
#connection_stringの生成
connection_string = "/dev/ttyACM0,115200"
 
# フライトコントローラ(FC)へ接続
print( "FCへ接続: %s" % (connection_string) )    # 接続設定文字列を表示
vehicle = connect(connection_string, wait_ready=True)    # 接続
 
def arm_and_takeoff(aTargetAltitude):
    """
    Arms vehicle and fly to aTargetAltitude.
    """
 
#    print("Basic pre-arm checks")
    # Don't try to arm until autopilot is ready
#    while not vehicle.is_armable:
#        print(" Waiting for vehicle to initialise...")
#        time.sleep(1)
 
    print("Arming motors")
    # Copter should arm in GUIDED mode
    vehicle.mode = VehicleMode("POSITION")
    vehicle.armed = True
 
    # Confirm vehicle armed before attempting to take off
    while not vehicle.armed:
        print(" Waiting for arming...")
        time.sleep(1)
 
    print("Taking off!")
    vehicle.simple_takeoff(aTargetAltitude) # Take off to target altitude
 
    # Wait until the vehicle reaches a safe height before processing the goto (otherwise the command
    #  after Vehicle.simple_takeoff will execute immediately).
    while True:
        print(" Altitude: ", vehicle.location.global_relative_frame.alt)
        #Break and return from function just below target altitude.
        if vehicle.location.global_relative_frame.alt>=aTargetAltitude*0.95:
            print("Reached target altitude")
            break
        time.sleep(1)
 
#高さ~mまで離陸
arm_and_takeoff(1)
 
#hovering5秒間
time.sleep(5)
 
print("Landing !")
vehicle.mode = VehicleMode("LAND")
 
#FCと接続を閉じる
vehicle.close()

[image]

Navron4500, 2023-08-25:
@mana747 Did you found the solution we are stuck exactly at the same problem. We are planning to use pymavlink instead of this what's you take on this?

sanjay-swain, 2024-02-10:
In the code you were putting the vehicle in `POSITION` mode. From ardupilot wiki POSITION mode.

It looks like you are using PX4 hardware. I don't think dronekit supports PX4. So that's probably is the issue