dronekit/dronekit-python #1171: gazebo virtual camera stream in dronekit-python script

AnaghBenjwal, 2022-11-11:
Is there a way to access the virtual camera that comes with the iris quadcopter in ardupilot gazebo SITL via a dronekit script?
Currently, we are using dronekit to send positional and velocity commands to the drone, but we also require access to the camera stream to implement basic computer vision / image processing on it.
I'm aware this is usually done using the libgazebo_ros_camera plugin while using ROS and mavros, but is there a way to access the virtual camera stream natively through dronekit-python?

If there isn't, is there a way I can access the camera stream via ROS while still running the dronekit script to send positional commands?

farshid-Rayhan, 2025-01-09:
any update on this ?

hamishwillee, 2025-01-19:
Probably not. YOu might ask on the project gitter channel.