dronekit/dronekit-python #1172: Unable to install dronekit on pi4, Missing depedancies

Radnartjie, 2022-11-11:
I'm having a bit of trouble installing python dronekit  on my raspberry pi 4B (8GB) (bullseye).

I know that dronekit does not support Python3, hence I am running a virtual python environement (dronekitty)

This is my terminal output:
I'm having a bit of trouble installing python dronekit  on my raspberry pi 4B (8GB) (bullseye).

I know that dronekit does not support Python3, hence I am running a virtual python environement (dronekitty)

What could i be doing wrong? I am pretty new to Linux.

```
pi@raspberrypi:~ $ source dronekitty/bin/activate
(dronekitty) pi@raspberrypi:~ $ pip install dronekit
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting dronekit
  Using cached dronekit-2.9.2-py2-none-any.whl (57 kB)
Collecting monotonic>=1.3
  Using cached monotonic-1.6-py2.py3-none-any.whl (8.2 kB)
Collecting pymavlink>=2.2.20
  Using cached pymavlink-2.4.37.tar.gz (5.7 MB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/dronekitty/bin/python /home/pi/dronekitty/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-IoOc4V/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=42' future lxml
       cwd: None
  Complete output (17 lines):
  DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=42
    Using cached setuptools-44.1.1-py2.py3-none-any.whl (583 kB)
  Collecting future
    Using cached future-0.18.2.tar.gz (829 kB)
  Collecting lxml
    Using cached lxml-4.9.1.tar.gz (3.4 MB)
      ERROR: Command errored out with exit status 1:
       command: /home/pi/dronekitty/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-OdNcuZ/lxml/setup.py'"'"'; __file__='"'"'/tmp/pip-install-OdNcuZ/lxml/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-qUR0KC
           cwd: /tmp/pip-install-OdNcuZ/lxml/
      Complete output (3 lines):
      Building lxml version 4.9.1.
      Building without Cython.
      Error: Please make sure the libxml2 and libxslt development packages are installed.
      ----------------------------------------
  ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/dronekitty/bin/python /home/pi/dronekitty/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-IoOc4V/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=42' future lxml Check the logs for full command output.
(dronekitty) pi@raspberrypi:~ $

RajendranDinesh, 2022-11-13:
Have You Tried Installing libxml2 and libxslt packages?

Lingdajin, 2023-01-26:
> How do i go about installing these two packages? When i try using sudo apt-get or pip, nothing is found?

try this "sudo apt-get install libxml2-dev libxslt1-dev zlib1g-dev python-py"
or you can try "sudo aptitude dronekit",this command can guide you to install dronekit.