dronekit/dronekit-python #1175: Stuck during connection to Pixhawk due to critical:autopilot: pream checks.

garrettadamsss, 2022-11-27:
Hello I am new to dronekit and am attempting to connect a pixhawk orange cube with Arduplane V4.3.1 Firmware with Jetson Nano. When running connection script with "connect('dev/ttyTHS1", baud=57600, wait_ready=true)", my terminal returns:
 "CRITICAL:autopilot:PreArm: AHRS: EKF3 Yaw inconsistent by 77 deg
CRITICAL:autopilot:PreArm: GPS 1: Bad fix
CRITICAL:autopilot:PreArm: Radio failsafe on"

The script then does not execute any other commands. TIA

hamishwillee, 2022-11-27:
Youve connected successfully but the vehicle has a bad fix. That means it can probably output some telemetry but it won't let you take off or anything. You'd get the same warnings from connecting to Mission Planner.

Probably your vehicle needs the compass calibrated. Connect to MIssion planner, see if you can get it properly calibrated, then it could well just work with Dronekit

closed 2022-11-27 by hamishwillee

garrettadamsss, 2022-11-29:
Thank you for the reply. I did fix all of the prearm checks. However I am still unable to connect. I ran a px4 under the same conditions and was able to connect. So there is an issue with the orange cube

hamishwillee, 2022-11-29:
That is possible. Unfortunately I can't help other than the high level "what does the debug indicate".