dronekit/dronekit-python #1178: Mission Planner and Dronekit simultaneously

hassaansidd, 2023-01-14:
I am planning to make a plane that can perform object detection and perform specific tasks while in flight. So for planning flights, I wanna use ardupilot mission planner and for object detection and for a certain movement of servos I wanna use dronekit. Is it possible to use both software partially...

kyrlon, 2023-04-06:
rough but I believe with mavproxy

saidinesh5, 2023-05-18:
Setup Mavlink Mirror with write access on Mission Planner. You can then make mission planner act as a proxy between your script and your plane.

```python
from dronekit import connect
vehicle = connect('udp:127.0.0.1:14550',baud=57600)
```

[image]