dronekit/dronekit-python #1182: dronekit-sitl not work with new version python.

orangefabri, 2023-02-13:
Hello.

it seems that dronekit-SITL not works with newer version python.

following error message orrcurs when I connect to dronekit-sitl. :

File "C:\Users\xxx\.conda\envs\xxx\lib\site-packages\dronekit\mavlink.py", line 127, in __init__
    if ip.startswith("udpin:"):
AttributeError: 'SITL' object has no attribute 'startswith'

--------------------

I changed all variable to string by str() but since I don't know about entire code structure, I failed.

kyrlon, 2023-02-18:
what version of python are you running?

orangefabri, 2023-02-18:
Im using latest python, 3.9.16

I remember I tried 3.9.10,3.9.0 but I'm not sure about the version I tried.

kyrlon, 2023-02-20:
try a virtual environment to make sure it is not a conflict with your system install:
```
PS C:\Users\kyrlon\Desktop> py -m venv env1
PS C:\Users\kyrlon\Desktop> .\env1\Scripts\activate
(env1) PS C:\Users\kyrlon\Desktop> py -m pip install dronekit-sitl
Collecting dronekit-sitl
  Using cached dronekit_sitl-3.3.0-py3-none-any.whl (38 kB)
Collecting psutil>=3.0
  Using cached psutil-5.9.4-cp36-abi3-win_amd64.whl (252 kB)
Collecting six>=1.10
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting dronekit>=2.0.0b6
  Using cached dronekit-2.9.2-py3-none-any.whl (56 kB)
Collecting monotonic>=1.3
  Using cached monotonic-1.6-py2.py3-none-any.whl (8.2 kB)
Collecting pymavlink>=2.2.20
  Using cached pymavlink-2.4.37-py3-none-any.whl (5.8 MB)
Collecting future
  Using cached future-0.18.3.tar.gz (840 kB)
Collecting lxml
  Using cached lxml-4.9.2-cp39-cp39-win_amd64.whl (3.9 MB)
Using legacy 'setup.py install' for future, since package 'wheel' is not installed.
Installing collected packages: lxml, future, pymavlink, monotonic, six, psutil, dronekit, dronekit-sitl
    Running setup.py install for future ... done
Successfully installed dronekit-2.9.2 dronekit-sitl-3.3.0 future-0.18.3 lxml-4.9.2 monotonic-1.6 psutil-5.9.4 pymavlink-2.4.37 six-1.16.0
WARNING: You are using pip version 21.2.3; however, version 23.0.1 is available.
You should consider upgrading via the 'C:\Users\kyrlon\Desktop\env1\Scripts\python.exe -m pip install --upgrade pip' command.
(env1) PS C:\Users\kyrlon\Desktop> py --version
Python 3.9.7
```

orangefabri, 2023-02-20:
> try a virtual environment to make sure it is not a conflict with your system install:
>
> ```
> PS C:\Users\kyrlon\Desktop> py -m venv env1
> PS C:\Users\kyrlon\Desktop> .\env1\Scripts\activate
> (env1) PS [...]

Thank you for your reply. Im already using virtual environment as you can see in my error message. do you run your dronekit sitl well in windows?

kyrlon, 2023-02-22:
I was able to install, but I have not been successful in running the sitl efficiently. I have been having a similar [issue](https://github.com/dronekit/dronekit-python/issues/1083) where after 10 secs, the sitl disarms the motors. The only solution is to downgrade to python3.6 and also install pymavlink==2.4.8, which both options I would prefer not to result to.

kyrlon, 2023-06-21:
I ended up using a docker container suggested from this [repo](https://github.com/Akademicki-Klub-Lotniczy/ardupilot-sitl-docker). I ended up just pulling the [image](https://hub.docker.com/r/wnt3rmute/ardupilot-sitl) directly

hamishwillee, 2023-06-21:
@peterbarker Is there any plan to update Dronekit Sitl to Python 3? It's a pretty useful tool.

ItayShiff, 2023-09-28:
Any update on this ?