dronekit/dronekit-python #1183: problem z dronekit

Pati2009xv3, 2023-03-10:
Traceback (most recent call last):
  File "D:\cheaty\Bocik\arduino 1.py", line 5, in <module>
    from dronekit import connect, VehicleMode
  File "D:\cheaty\Bocik\Bocik\Lib\site-packages\dronekit\__init__.py", line 2689, in <module>
    class Parameters(collections.MutableMapping, HasObservers):
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableMapping'

astro-phil, 2023-04-27:
Hi, my work around for this was:

open your file: "D:\cheaty\Bocik\Bocik\Lib\site-packages\dronekit_init_.py"
replace "import collections" with "import collections.abc as collections"

Hope this fixes your error.
Best regards

Phil

morzack, 2023-06-28:
Just for a bit of semi-closure on this: it looks like a commit was merged into master a while back that should solve this issue (1d89e82) *however* there has not been a pip/pypi release since then.

For people coming to this issue in the future, you might have luck by installing dronekit directly from source (until the next release...):

`pip install git+https://github.com/dronekit/dronekit-python.git`

hamishwillee, 2023-06-29:
That's great to know. @morzack Are you planning a release, and do you have enough information to do so? 
It might be good to automate this in CI too.

I have never done one of this project. If you don't have enough information, we can ask Peter.

morzack, 2023-07-01:
@hamishwillee I suspect that we should make a patch release at some point once the most common issues have been resolved. I'm not entirely sure how to do that (and would defer to Peter regardless of if I knew how :) )

darshit-desai, 2023-07-07:
@morzack I tried installing using above command on windows, (all of python toolchain is already there) but it gives me an error: 

```shell
C:\Users\darshit>pip install git+https://github.com/dronekit/dronekit-python.git
Collecting git+https://github.com/dronekit/dronekit-python.git
  Cloning https://github.com/dronekit/dronekit-python.git to c:\users\darshit\appdata\local\temp\pip-req-build-ri7slywv
  Running command git clone --filter=blob:none --quiet https://github.com/dronekit/dronekit-python.git 'C:\Users\darshit\AppData\Local\Temp\pip-req-build-ri7slywv'
  Resolved https://github.com/dronekit/dronekit-python.git to commit 3ee89b7d83ba1fe676f0b2980a7774b052c29c2f
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [10 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\darshit\AppData\Local\Temp\pip-req-build-ri7slywv\setup.py", line 7, in <module>
          LongDescription = f.read()
                            ^^^^^^^^
        File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1264.0_x64__qbz5n2kfra8p0\Lib\encodings\cp1252.py", line 23, in decode
          return codecs.charmap_decode(input,self.errors,decoding_table)[0]
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      UnicodeDecodeError: 'charmap' codec can't decode byte 0x8f in position 1037: character maps to <undefined>
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

``` 

I will try the same command on a fresh ubuntu install

morzack, 2023-07-07:
Hmm, I wonder if this has anything to do with the "⚠️" that was recently added to the README. It looks like `setup.py` reads that file and so the encoding used for the read in windows might be unhappy with that character.

This isn't entirely conjecture -- I did a quick bisect to check and the commit that updated the README seems to be the bad one.

I assume that we can can either change `setup.py` to use the correct encoding on read, or update `README.md` to not use that character.

cc @darshit-desai

darshit-desai, 2023-07-07:
Ok so the command `pip install git+https://github.com/dronekit/dronekit-python.git` did work on ubuntu. I guess I installed the dronekit before using `pip install dronekit` from the docs and than removed it after encountering this issue #1183. Followed the above command and it didn't work for windows.