dronekit/dronekit-python #1184: command in real time

x123y123, 2023-04-06:
Hi, I want to ask question about "if I already send a command(eq, vehicle.takeoff(5))", can it be interrupt(eq, maybe when it goes up to 3 meters be interrupted) or cover by new command(eq, vehicle.takeoff(7))?

ShafiqSadat, 2023-11-14:
In the context of DroneKit, interrupting or overriding a command depends on how the underlying flight controller and autopilot system handle such commands. The ability to interrupt or override a command might vary based on the specific autopilot software (e.g., ArduPilot) and the vehicle's firmware.

In the case of the vehicle.takeoff(5) command, if you want to interrupt or override it when the drone reaches a certain altitude (e.g., 3 meters), you would typically need to implement logic in your code to check the current altitude and send a new command accordingly.

Here's a simplified example in Python:

```import time

# Assuming 'vehicle' is your DroneKit vehicle instance

# Send the initial takeoff command
vehicle.simple_takeoff(5)

# Wait for the vehicle to reach 3 meters
while vehicle.location.global_relative_frame.alt < 3:
    time.sleep(1)

# Send a new takeoff command to change the target altitude to 7 meters
vehicle.simple_takeoff(7)
```

In this example, the code waits until the drone reaches an altitude of 3 meters and then sends a new takeoff command to change the target altitude to 7 meters. This is a simplistic example, and in a real-world scenario, you might want to implement more robust error handling and safety checks.

Keep in mind that the specifics may vary depending on the autopilot system, vehicle type, and firmware version. Always refer to the documentation of the specific autopilot system you are using for accurate and up-to-date information.

hamishwillee, 2023-11-14:
Correct. Flight stacks will reject a second takeoff command if the first one is still being handled (i.e. has not yet returned an ACK to say it has been accepted). After that, the behaviour is flight stack dependent - it might accept a second takeoff point and just treat it as a waypoint. In the above, you might as well just set a new target to go to as set another takeoff.

closed 2023-11-14 by hamishwillee