dronekit/dronekit-python #1185: Is this project still active?

krishnaTORQUE, 2023-04-07:
Is this project still active?

hamishwillee, 2023-04-07:
Not very. But it has a busy community and is not allowed to get too broken with respect to working with ArduPilot.

YacobBY, 2023-05-28:
Why is the project deprecated? It's not working well with newer python versions anymore so unless updates will be done this rather popular project just exists to generate confusion for newbies.

hamishwillee, 2023-05-28:
Because the people who developed it were paid, and then no longer paid. There are alternatives like MAVSDK that are better supported, but are tested against PX4. There is ad-hoc support from various users and developers including many in the ArduPilot community, but not official support.

Upshot, there is no one who owns this, but enough people use it and can keep it going.

If you discover a specific problem post an bug report and cross post the the gitter channel for discussion. That is the best way to get it addressed.

YacobBY, 2023-05-30:
@hamishwillee Thanks for the reply. My (and others) problem is mostly that it's very unclear which projects are actively maintained causing us to lose a lot of time having to sieve through all the available options. It would be helpful if the top of the readme just stated that this project is deprecated and will not work with modern python interpreters anymore.

I know that it's a ungrateful to all the hard work that you open source maintainers do, but it's genuinely very time-consuming and frustrating that all these library "choices" ultimately lead to nowhere instead of just a bright red arrow pointing to mavsdk-python. I have lost at least 6 hours testing stuff only to find out that you just can't get it to run without downgrading my OS to Ubuntu 18.04 and running Python3.6 which is not really something I would like to restrict the rest of my software stackt o.

hamishwillee, 2023-06-01:
Fair enough. @peterbarker @mrpollo It would be good to make a support statement about this project - something along the lines of: 

> This project does not have an active maintenance team, but does have significant community support and use.
> For support options see: [Link] (probably gitter)
> Applicants with Python and DroneKit experience are welcome to apply for a maintainer position [link where?]

mrpollo, 2023-06-07:
Maintainers are welcomed, please apply by creating a GitHub issue and listing the reasons why you want to help, and we will be granting committer access to folks who are interested in helping. 

Does this help?

hamishwillee, 2023-06-08:
I installed on Ubuntu 20.04 LTS that comes with Python 3.8 using the https://dronekit-python.readthedocs.io/en/latest/contributing/developer_setup_linux.html instructions (but for Python3). I also installed ArduPilot using current structions.

I then modified the helloworld in the getting started to use the connection string `127.0.0.1:14551`, use `print()` (not necessary) and disable dronekit-sitl (not sure if that runs on Python 2). This worked fine:

```
#print "Start simulator (SITL)"
#import dronekit_sitl
#sitl = dronekit_sitl.start_default()
#connection_string = sitl.connection_string()

# Import DroneKit-Python
from dronekit import connect, VehicleMode

#connection_string = 'tcp:127.0.0.1:5760'
connection_string = '127.0.0.1:14551'

# Connect to the Vehicle.
print("Connecting to vehicle on: %s" % (connection_string,))
vehicle = connect(connection_string, wait_ready=True)

# Get some vehicle attributes (state)
print("Get some vehicle attribute values:")
print( " GPS: %s" % vehicle.gps_0)
print(" Battery: %s" % vehicle.battery)
print( " Last Heartbeat: %s" % vehicle.last_heartbeat)
print( " Is Armable?: %s" % vehicle.is_armable)
print( " System status: %s" % vehicle.system_status.state)
print( " Mode: %s" % vehicle.mode.name )    # settable

# Close vehicle object before exiting script
vehicle.close()

# Shut down simulator
#sitl.stop()
print("Completed")
```

I then ran the mission_basic example here: https://dronekit-python.readthedocs.io/en/latest/examples/mission_basic.html

That worked fine too:
```
python3 mission_basic.py --connect '127.0.0.1:14551'
Connecting to vehicle on: 127.0.0.1:14551
Create a new mission (for current location)
 Clear any existing commands
 Define/add new commands.
 Upload new commands to vehicle
Basic pre-arm checks
Arming motors
 Waiting for arming...
Taking off!
 Altitude:  0.002
 Altitude:  0.002
 Altitude:  0.002
 Altitude:  0.13
 Altitude:  2.036
 Altitude:  4.706
 Altitude:  6.936
 Altitude:  8.519
 Altitude:  9.561
Reached target altitude
Starting mission
Distance to waypoint (0): None
Distance to waypoint (1): 79.12361098829477
Distance to waypoint (1): 78.52048617347582
Distance to waypoint (1): 76.47935934706021
Distance to waypoint (1): 72.25455364740925
Distance to waypoint (1): 65.62748730820536
Distance to waypoint (1): 56.7901708625966
Distance to waypoint (1): 46.58611471205202
Distance to waypoint (1): 35.83533114082336
Distance to waypoint (1): 25.11513014056162
Distance to waypoint (1): 15.261418888859415
Distance to waypoint (1): 7.316937665009313
Distance to waypoint (1): 2.2171961091584462
Distance to waypoint (2): 121.1800861210931
Distance to waypoint (2): 117.44100860042782
Distance to waypoint (2): 110.56095755152107
Distance to waypoint (2): 101.42109877648008
Distance to waypoint (2): 90.72288898673054
Distance to waypoint (2): 79.26811001667626
Distance to waypoint (2): 67.55722851505368
Distance to waypoint (2): 55.72394782858215
Distance to waypoint (2): 43.72371470581943
Distance to waypoint (2): 31.6677941054081
Distance to waypoint (2): 23.0294095078316
Distance to waypoint (2): 12.899359772946926
Distance to waypoint (2): 5.3749532479888105
Distance to waypoint (2): 1.4969698710149022
Distance to waypoint (3): 95.88809350297983
Skipping to Waypoint 5 when reach waypoint 3
Distance to waypoint (5): 153.2892582649245
Exit 'standard' mission when start heading to final waypoint (5)
Return to launch
Close vehicle object
```

hamishwillee, 2023-06-08:
@YacobBY As per ^^^ this works for me on a valid version of Python (3.8) with very minor modifications. It may be that you're using the wrong/incorrectly documented string, or that there other parts I am not testing that are failing. I'd be interested in knowing what you are finding fails before adding a note that says it doesn't work.

@mrpollo I still think it worth having a note that we need more maintainers. It would also be well worthwhile updating all the instructions for Python 3 and Ubuntu 20.02 or 22.02.

mrpollo, 2023-06-08:
hey @hamishwillee thanks for the ping, I agree with your assessment, Python 3 support exists but its not a good experience for everyone. If anyone is interested in helping please apply to the position.

YacobBY, 2023-06-08:
@hamishwillee Thanks for updating the Readme.

I don't want to start a debate about it but here's my newcomer experience:

1) Open this project at https://github.com/dronekit/dronekit-python
2) Click the "Quickstart" link
3) Do the install stuff at the top of the quickstart
4) Try to run the "Basic “Hello Drone”" script 
5) The Script still uses Python 2 syntax for printing and is not even Python 3 compatible

I'm sure I could easily get this to work using the adjustments you provided, however a less experienced developer that just wants to start experimenting with drones would not have this knowledge and it would cost them a lot of time. A quickstart guide should allow a newcomer to rapidly start using a project, especially since Dronekit was made to simplify drone progarmming.

In any case I have since moved on to using MavSDK-Python. This does have slightly more complex syntax and looks less appealing than dronekit but the provided examples and tutorials do (mostly) work. The fragmentation between all these different drone programming libraries doesn't seem to add to the developer experience though.

hamishwillee, 2023-06-08:
@YacobBY Absolutely agree.

This is a community supported project now, so if you'd like to contribute fixes like that you'd be most welcome. Or go on the gitter channel and ask other members to help. I might do it at some point, but I'm no longer working in this project and have far too much to do that people will pay me for. I would take the time to review someone elses' doc updates though.

I do like Dronekit-Python because I can extend it myself with only limited Python skills. But MAVSDK is a much better supported project, so IMO so if I had to work in Python I would either use that or Pymavlink.

mrpollo, 2023-06-09:
Another great option is https://github.com/Auterion/libmav-python

hamishwillee, 2023-06-09:
> Another great option is https://github.com/Auterion/libmav-python

Great potential, but a long way to go. It's a message sender/receiver with no microservice support, which you would have to implement yourself. Also docs quite limited. 
I'm going to have a play with it sometime.

hamishwillee, 2023-07-06:
@morzack Is co-ordinating review and such now. Closing this as "starting to be more active again".

closed 2023-07-06 by hamishwillee