dronekit/dronekit-python #1189: Dronekit Servo Control

umerkh78, 2023-06-02:
Hello everyone. I'm new to dronekit and I'm trying to connect my LEDs to servo outpus of Pixhawk 2.4.1.
Does dronekit provide any support for the servo outputs or not?

morzack, 2023-06-28:
Hi, as far as I understand, there is not first-class support for servos in dronekit (helper functions, etc). However, you can definitely control servo channels by sending a [`MAV_CMD_DO_SET_SERVO`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO) command.

Something like the snippet below could help:
```python
import dronekit
from pymavlink import mavutil

vehicle: dronekit.Vehicle

msg = vehicle.message_factory.command_long_encode(
        0, 0, # target system, component
        mavutil.mavlink.MAV_CMD_DO_SET_SERVO, 0,
        servo_channel, pwm,
        0, 0, 0, 0, 0)
vehicle.send_mavlink(msg)
```

hamishwillee, 2023-06-29:
Thanks @morzack. Just to be pedantic, support for any particular message depends on the flight stack and version too (a first-class solution in DroneKit might abstract that). 

For example, in PX4 you would send https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR and you **might** also have to configure the actuator outputs used.

hamishwillee, 2023-06-29:
Closing this as answered. Thanks @morzack

closed 2023-06-29 by hamishwillee