dronekit/dronekit-python #1192: Expressing interest in maintaining dronekit

morzack, 2023-06-14:
Hi again. I recently saw the request for maintainers go out and felt like it would be worth mentioning my interest in contributing.

I’m currently associated with a research project at a university that has been using dronekit extensively for several years now. One of the earliest tasks I had on that project was to write an asyncio powered wrapper for dronekit because of the growing difficulty of writing complex dronekit scripts! (You can find a slightly out of date version of this wrapper at https://github.com/morzack/aerpawlib-vehicle-control/). As a team, we also work with bleeding edge versions of ardupilot/various COTS hardware and semi-regularly must iterate on ways to make dronekit compatible with them. A recent example of a dronekit-related bug that we’ve encountered and proposed a patch for is #1187.

Personally, I hope to use my experience with Python and dronekit to help make sure that this library remains functional for users. While I understand that projects like mavsdk continue to grow in importance and use (and reasonably should), dronekit is still a critical piece of infrastructure for many teams that deserves attention. My general goals and involvement would be centered around providing support for the incoming issues+PRs and helping modernize some of the critically out of date code (specifically by reducing Python 2 dependency, especially in documentation).

hamishwillee, 2023-06-15:
@morzack That sounds really well aligned with what needs to be done to keep this project alive. I'm particularly keen on moving to a solid Python 3 implementation as soon as possible, and making Python 2 deprecated.

First things first, how about you join as a developer https://github.com/orgs/dronekit/teams/developers - that will allow you to start reviewing PRs. Perhaps in the first instance CC in @peterbarker with your opinion so that we can make sure that fixes are aligned with existing goals.

When you start your own PRs we'll need to get @peterbarker to review them. Mostly to make sure that we have test processes in place, and because it's a lonely place to be without independent eyes. 

Does that make sense?

hamishwillee, 2023-06-15:
PS He's merged your PR!

hamishwillee, 2023-06-15:
@mrpollo can you please add @morzack to the Dronekit project, developer team? I can't work out how to do it.

mrpollo, 2023-06-16:
Hey @morzack thanks for taking the time, I just went ahead and added you as a team member. Please use your new powers responsibly. Let us know if you need anything!

morzack, 2023-06-20:
Thanks -- closing this now, I'll run through the PRs and issues at some point in the near future (probably in ~2 weeks, currently finishing up finals @ uni).

closed 2023-06-20 by morzack

hamishwillee, 2023-06-21:
Thanks @morzack . Good luck at finals.

mrpollo, 2023-06-22:
Thanks @morzack please let us know if you need anything! I have access to domains, and servers if you need to change anything for docs or any infrastructure needs.