dronekit/dronekit-python #1195: Can I use sbus in dronekit?

chano1025, 2023-06-22:
Previously, I gave a pwm signal to the servo motor to control the Viewpro gimbal camera. 
However, I want to convert it to the sbus but I'm not sure how to use it.
Is sbus also controlled with Mavlink messages like servo motors?

closed 2023-06-22 by hamishwillee

chano1025, 2023-06-23:
Thank you for answer.
Is there a way to transfer to a port such as uart, telem, or i2c?

hamishwillee, 2023-06-23:
YOu send MAVLink commands via some radio or IP serial link to do "stuff". Those commands get turned internally into some operation on the flight stack.

MAVLink provides an interface to set arbitrary actuators (or servos) on the main outputs of a flight stack - i.e. that means UAVCAN or PWM.  There is no way to send an arbitrary command to other ports via MAVLink. 

To use a gimbal, MAVLink provides the gimbal API. It would be up to a flight stack to decide how the hardware works when it gets one of those commands. Currently flight stacks attach to PWM outputs or a "pure mavlink" for gimbals. So while in theory you might control a gimbal on those other busses you would have to write a driver and gimbal backend to do the work.

So it is kind of "not a dronekit or mavlink question".