dronekit/dronekit-python #1196: Interested in becoming a maintainer

darshit-desai, 2023-07-06:
Hi developer team,

I am interested in becoming a contributor for this repository and here are my experiences with dronekit, Ardupilot and PX4 which helps me better understand the improvements which might come up in the future,

I have used dronekit python earlier, to develop an airdrop selection script which used data streamed (airspeed, ground speed, orientation) through mavlink to actuate the servo of the bottle drop.
I have used Mavros and PX4 to implement an offboard obstacle avoidance script, which takes data from d435i stereo camera and orients the drone away from the obstacle based on the pose of the obstacle.
I have also worked with the core hardware and know about the communication protocols of the RC hardware.

Thank you for considering me,
Best,
Darshit

hamishwillee, 2023-07-06:
Thanks @darshit-desai 

@morzack is volunteered recently too, and is co-ordinating reviews of PRs and issues. Please collaborate. I would like you to work together on a few reviews and also co-ordinate with @peterbarker before giving you merge rights.

darshit-desai, 2023-07-06:
Thank you @hamishwillee, I sent an email to @morzack. Is there a place where I can get started, I come from the user side where I only used Dronekit as an API to write apps and not really know what actually goes on under the hood.

morzack, 2023-07-07:
Hi, I just replied!

Just for transparency here, I mentioned how I've been going through the open issues and categorizing them into several buckets (stale, urgently needs fixing, etc...).

@hamishwillee in your opinion is reviewing the currently open PRs the most relevant first step (as opposed to any dev work on issues)?

hamishwillee, 2023-07-07:
@morzack My opinion is that getting DroneKit rock solid in Python 3 should be the "project priority".

But mostly I want to keep the project alive and lively, so what I'm after is professionalism, process, common sense. As long as we see that, it should be the priorities of the people actually doing the work that is most important - i.e. you.

TongBingda, 2023-07-12:
Hi, developer team of dronekit project,

I'm a developer from China, I've added some new features in the last few days and I'm interested in becoming a contributor to this repository. I am currently a PhD student. I have been working with dronekit and ardupilot for UAS development since 2020, which helps me to better participate in future updates of the project.I am currently using drone python to develop a lightweight drone cluster ground station script that obtains drone information and controls drone state by using the pysimplegui library and the dronekit library. I have also used a Raspberry Pi and NVIDIA jetson NX as companion computers to control quadcopters and fixed wing drones. I have also used remote controllers like the radiomaster tx16s that have openTX or edgeTX open source operating systems installed.

I have emailed @morzack . Thank you for considering me!

morzack, 2023-07-12:
Hi @TongBingda , feel free to make a pull request with the changes and then it can be tested + reviewed

hamishwillee, 2023-08-08:
So any progress on this?. Would be good to see some PRs for concrete Python 3 support.

hamishwillee, 2025-07-30:
We've had a number of people express interest in becoming maintainers -  @morzack @a87chaud @alymkarim @Ian-VA @Russi1555 @Tuxliri  (related discussions, which I will close, are #1240, #1237, #1228, #1225).

To be clear there are no maintainers currently - I used to work on the documentation but I have moved on. This is still used by the ArduPilot community. 

The next main steps required are to get this working **solidly** with Python 3, make a release, and update the documentation. 

In the meantime I would suggest that if you are interested in helping you start off my looking at existing issues in pairs or more and see if you can agree on fixes. Once we have a few people with some proven expertise and ongoing commitment we can move forward on a solid base.

alymkarim, 2025-07-30:
> We've had a number of people express interest in becoming maintainers - [@morzack](https://github.com/morzack) [@a87chaud](https://github.com/a87chaud) [@alymkarim](https://github.com/alymkarim) [...]

yep that sounds good to me, should we start to work on it? @morzack  @a87chaud @Ian-VA @Russi1555 @Tuxliri

a87chaud, 2025-07-31:
Hi @alymkarim  I would be happy to work on any issue.
@hamishwillee is there any critical issue we should start off with? it seems there is security issue #1239 which we could start off with.

hamishwillee, 2025-07-31:
I think you should work together to find issues you think are important. Personally I think that security issue is low priority. My reasoning is that very few people use IP endpoints in dronekit but many use everything else.

I consider any bugs that prevent this running on Python 3.7 or the latest python (whatever that is) to be the important ones. Once we have fixed all those we'd be in a reasonable position to cut a new release.

a87chaud, 2025-07-31:
There doesnt seem to be support for sitl for python3 would that be a good starting point?

a87chaud, 2025-07-31:
Cool! I will read the documentation and get more familiar with it. Any tips/resources on how to get started?

Optimized-Brain, 2025-09-22:
Hi @hamishwillee , I am also interested in contributing to this project. Please consider.