dronekit/dronekit-python #1199: DroneKit using any flight controller

Giol-Vinyals-i-Roca, 2023-07-27:
Is it possible to use drone kit with a Raspberry Pi and communicate to a “normal” (non ardupilot) flight controller, via serial?

Shreyas-dotcom, 2023-08-03:
DroneKit should be compatible with any flight controller that uses the Mavlink flight communications protocol. Keep in mind though - if you're using a non ardupilot FC, there's a good chance some of the features may not work, and you'd have to edit some of the code to get it working.

hamishwillee, 2023-08-03:
Yes-ish.

MAVLink is supposed to be flight-stack agnostic but the fact is that there are differences between flight stack implementations even in core services such as parameter exchange and mission upload. For example, ArduPilot keeps the home location in the zeroth item of a mission, while PX4 does not, and if  working with raw MAVlink, you need to be aware of that.

DroneKit is "somewhat aware" of the differences between ArduPilot and PX4 so to some extent you can use it for either. However it was developed/tested against ArduPilot. That means that you can use it with PX4 but it isn't as well tested. If you're using some other flight stack it could be even less useful.

If working with PX4 I'd recommend MAVSDK - https://mavsdk.mavlink.io/main/en/index.html

closed 2023-08-03 by hamishwillee