dronekit/dronekit-python #1201: problem with the heartbeat

naorwaiss, 2023-08-09:
/home/fhsbs/.virtualenvs/pythonProject2/bin/python /tmp/pycharm_project_173/test.py 
WARNING:dronekit:Link timeout, no heartbeat in last 5 seconds

i onley try to start this code:

rom dronekit import connect, VehicleMode
import time

# Connect to the simulator running on the local machine
connection_string = "/dev/ttyTHS1"
vehicle = connect(connection_string, wait_ready=True)

def check_armable():
    while not vehicle.is_armable:
        print("Waiting for GPS and armability...")
        time.sleep(1)

    print("Vehicle is armable")

def main():
    check_armable()

    # The rest of your code here
    # arm_and_takeoff(target_altitude)
    # ... other code ...

if __name__ == "__main__":
    main()

i use python 3.9.12 at my jetson nano 
dronekit               2.9.2
dronekit-sitl          3.3.0
exceptiongroup         1.1.2
future                 0.18.3
gpxpy                  1.5.0
idna                   3.4
iniconfig              2.0.0
jmespath               1.0.1
lxml                   4.9.3
marshmallow            3.17.0
MAVProxy               1.8.65

another things... i try to activate the simulation - dronekit-sitl but it didnt gave me to start the simulation...
mabey i need another python verson ??

ShafiqSadat, 2023-11-14:
Use ArduPilot SITL for simulation!

sim_vehicle.py -v ArduCopter --out=udp:192.168.1.12:14551

and for dronekit 

connection_string = "udpin:0.0.0.0:14551"
vehicle = connect(connection_string, wait_ready=True)

hamishwillee, 2023-11-14:
Thanks @ShafiqSadat - marking this as closed.

That said, this may also be a PYthon issue. Support for Python 3 is only very basic.

closed 2023-11-14 by hamishwillee