dronekit/dronekit-python #1202: Sending Velocity and Yaw command in same message in guided mode

pd3a, 2023-09-16:
Hi, Can I send both velocityned and yaw command simultaneously in same MAVLINK message either from global frame or relative frame? If yes what typemask should i use?

hamishwillee, 2023-09-17:
You mean in https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_GLOBAL_INT ? Maybe - it really would depend on your flight stack, which you have not stated. I think you'd have to look at the source code of that stack to be sure. Best place to ask might be on the Flight stack's support channel.

pd3a, 2023-09-17:
I'm using ArduPilot. What would be the typemask for sending just velocity and yaw commands and igonre others?

hamishwillee, 2023-09-17:
I'm not going to work that out for you - but you can do it yourself from https://mavlink.io/en/messages/common.html#POSITION_TARGET_TYPEMASK

pd3a, 2023-09-26:
I used the right bitmask which accepts only ned speeds and yaw '0b0000101111000111' in position global int frame, velocities commands are working **but yaw is not changing**. It's being ignored. can someone please help me here?

hamishwillee, 2023-09-26:
I suggest you ask on the ArduPilot forums. If you are sending the right message then it's probable that the flight stack doesn't support that combination.

jjshoots, 2023-09-29:
Did you manage to solve this? I'm using the github version of dronekit with Python 3.10 and Ardupilot 4.4, it accepts takeoff and land commands and performs the takeoff and land properly, but for some reason it keeps ignoring my mavlink messages of `set_position_target_local_ned`, the drone just hovers there and does nothing, no matter what kind of bit mask I use. My command is constructed as so:

```python
    def update_velocity_setpoint(self, frdy: np.ndarray) -> None:
        """Sets a new velocity setpoint.

        Args:
            frdy (np.ndarray): frdy
        """
        setpoint = self.enu2ned(frdy)

        # check the flight ceiling, downward is positive
        vehicle_height = self.vehicle.location.global_relative_frame.alt
        if vehicle_height > self.flight_ceiling:
            setpoint[2] = min(vehicle_height - self.flight_ceiling, 1.0)
        elif vehicle_height < self.flight_floor:
            setpoint[2] = max(vehicle_height - self.flight_floor, -1.0)

        self.setpoint_msg = self.vehicle.message_factory.set_position_target_local_ned_encode(
            # time_boot_ms (not used)
            0,
            # target system, target component
            0,
            0,
            # frame
            mavutil.mavlink.MAV_FRAME_BODY_OFFSET_NED,
            # type_mask, addressed in reversed, and used to indicate which components should be IGNORED
            # bit1:PosX, bit2:PosY, bit3:PosZ, bit4:VelX, bit5:VelY, bit6:VelZ, bit7:AccX, bit8:AccY, bit9:AccZ, bit11:yaw, bit12:yaw rate
            0b010111000111,
            # x, y, z positions (not used)
            0,
            0,
            0,
            # x, y, z velocity in m/s
            setpoint[0],
            setpoint[1],
            setpoint[2],
            # x, y, z acceleration (not used)
            0,
            0,
            0,
            # yaw, yaw_rate (only yaw rate used)
            0,
            setpoint[3],
        )
```

And I send the command simply via `vehicle.send_mavlink()`, once per second. Any ideas?

pd3a, 2023-09-30:
Hi, I was able to send only velocity inputs, yaw is still ignored even after changing the bitmask. But I changed my velocity vector using a transformation matrix. But in your message, as per mavlink documentation MAV_FRAME_BODY_OFFSET_NED is replaced by MAV_FRAME_BODY_FRD. Try that out, i think that should work.

hamishwillee, 2023-10-01:
Possibly ... you need to be in offboard mode, and perhaps already streaming before going into offboard mode.

IsacSmile, 2023-11-28:
from pymavlink import mavutil

# Connect to the autopilot (replace 'udp:127.0.0.1:14550' with your connection string)
master = mavutil.mavlink_connection('udp:127.0.0.1:14550')

# Set velocity and yaw command
vx = 2.0  # desired velocity in x-axis
vy = 0.0  # desired velocity in y-axis
vz = 0.0  # desired velocity in z-axis
yaw_rate = 45.0  # desired yaw rate in degrees per second

# Create a SET_POSITION_TARGET_LOCAL_NED message
msg = master.mav.set_position_target_local_ned_encode(
    0,  # time_boot_ms
    1,  # target system
    1,  # target component
    mavutil.mavlink.MAV_FRAME_LOCAL_NED,  # frame
    0b0000111111000111,  # type_mask
    0,  # x
    0,  # y
    0,  # z
    vx,  # vx
    vy,  # vy
    vz,  # vz
    0,  # afx (acceleration in x)
    0,  # afy (acceleration in y)
    0,  # afz (acceleration in z)
    0,  # yaw
    0,  # pitch
    yaw_rate,  # yaw_rate
)

# Send the message
master.mav.send(msg)