dronekit/dronekit-python #1207: How can I get connection strings when connect multi vehicles by Radio?

Residualstress, 2023-10-06:
I  have problem in building the connection to multi vehicles as I'm involved in a swarm project. Here is my  situation:

1) I have 3 holybro P900 radio and set them into Mesh mode with one is master and two are slaves.
2)The master radio connect with my laptop by USB when two slave radios connect with two Pixhawk by telem1 port

What I want is to connect to this two vehicles by Drone-kit python package and do some actions in my own python scripts. And actually, I know how to use "connect()" function in Drone-kit package such as vehicle = connect('udp:127.0.0.1:14550', baud=57600, wait_ready=True).

For my situation, how can I get the connection strings? I want to connect this two Pixhawk by vehicle_1= connect(xxxx) and vehicle_2 = connect(xxx). Should I use UDP port as well? And how can I get the UDP address of this two vehicles ?

Looking forward to your reply!

Weijie

ShafiqSadat, 2023-11-14:
You can use mavproxy to broadcast on UDP ports. For example:

mavproxy.py --master=COM5 --out=udp:192.168.1.12:14551

and

mavproxy.py --master=COM6 --out=udp:192.168.1.12:14552

For DroneKit, you can connect to each vehicle as follows:

vehicle1 = connect('udpin:0.0.0.0:14551', wait_ready=True)
vehicle2 = connect('udpin:0.0.0.0:14552', wait_ready=True)

hamishwillee, 2023-11-14:
Thanks @ShafiqSadat - that's great. If we get an answer back from @Residualstress that this answers the question, it might be worth adding something to the documentation showing how to address this case.