dronekit/dronekit-python #1212: Failing to arm motors

swere10, 2023-12-14:
-PC:~/Downloads$ python3.8 simple_goto.py
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
CRITICAL:autopilot:APM:Copter V3.3 (d6053245)
CRITICAL:autopilot:Frame: QUAD
CRITICAL:autopilot:Calibrating barometer
CRITICAL:autopilot:Initialising APM...
CRITICAL:autopilot:barometer calibration complete
CRITICAL:autopilot:GROUND START
Basic pre-arm checks
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
 Waiting for vehicle to initialise...
Arming motors
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...
 Waiting for arming...

swere10, 2023-12-14:
please help me out

hamishwillee, 2023-12-14:
Please raise this on the gitter or ArduPilot specific forums. Note, the errors tell you that you need to calibrate your vehicle. It won't take off until it is safe.

closed 2023-12-14 by hamishwillee