dronekit/dronekit-python #1216: Yaw don`t updated properly

TattooKot, 2024-01-30:
My drone in set up correclty.
First part of script works as expexted:

```
   print("Take off command received")
    # Arm the drone
    vehicle.armed = True
    while not vehicle.armed:
        time.sleep(1)

    # Take off to an altitude of 10 meters
    takeoff_altitude = 10.0
    # vehicle.wait_simple_takeoff(takeoff_altitude)
    vehicle.simple_takeoff(takeoff_altitude)

    # Wait for the takeoff to complete
    while True:
        if vehicle.location.global_relative_frame.alt >= takeoff_altitude * 0.95:
            print("Takeoff complete")
            break
        time.sleep(1)
```
But right after if I am trying to rotate drone, and nothing is happening.  

```    if relative:
        is_relative = 1 #yaw relative to direction of travel
    else:
        is_relative = 0 #yaw is an absolute angle
    # create the CONDITION_YAW command using command_long_encode()
    msg = vehicle.message_factory.command_long_encode(
        0, 0,    # target system, target component
        mavutil.mavlink.MAV_CMD_CONDITION_YAW, #command
        0, #confirmation
        heading,    # param 1, yaw in degrees
        0,          # param 2, yaw speed deg/s
        1,          # param 3, direction -1 ccw, 1 cw
        is_relative, # param 4, relative offset 1, absolute angle 0
        0, 0, 0)    # param 5 ~ 7 not used
    # send command to vehicle
    vehicle.send_mavlink(msg)
```    
   Maybe i miss something?
   Or maybe some one rotate drone using other commands?
   Please help.
   Thanks!

hamishwillee, 2024-01-30:
Generally speaking you should add logging to check that your command is actually getting called, and monitor the returned CMD_ACK to confirm what the flight stack said on receiving the message. That will tell you if the message is supported - support for any command depends on the particular flight stack and may also depend on flight mode and parameter values.

I'd raise this in the gitter channel: https://app.gitter.im/#/room/#dronekit_dronekit-python:gitter.im - because this is probably not an issue with dronekit specifically but your usage.

closed 2024-01-30 by hamishwillee

TattooKot, 2024-01-30:
@hamishwillee thanks, will do!
Ps. I am using dronekit-silt on GUIDED mode, so I'm sure that it not drone issue. I also can't change mode from code for some reason, but I saw oped issue for this also

hamishwillee, 2024-01-30:
Cool. This repo is not well maintained - you will definitely get more help from the community.