dronekit/dronekit-python #1220: Dronekit Python QuadCopter Kamikaze Drone Pitch Settings

Muhammetyilmaz7, 2024-05-04:
Hello, I am using dronekit python in quadcopter drone. This drone will perform a task autonomously. Let me briefly talk about the mission: The drone takes off and heads towards the specified location. After reaching a certain distance on the road, it must go to that location in a kamikaze dive manner. I am having trouble adjusting the pitch of the drone. How can I do this? How can the nose of the drone move to the desired position at a desired angle?

captain-woof, 2024-06-01:
I smell something fishy here

nicewrld, 2024-06-01:
This is completed? What did you do to solve your issue @Muhammetyilmaz7? I would love to hear more about the outcome.

Muhammetyilmaz7, 2024-06-01:
> Bu tamamlandı mı? Sorununuzu çözmek için ne yaptınız?@Muhammetyilmaz7? Sonuç hakkında daha fazla şey duymayı çok isterim.

Unfortunately I couldn't find a solution :(

Muhammetyilmaz7, 2024-08-13:
Hello, I am using dronekit python in quadcopter drone. This drone will perform a task autonomously. Let me briefly talk about the mission: The drone takes off and heads towards the specified location. After reaching a certain distance on the road, it must go to that location in a kamikaze dive manner. I am having trouble adjusting the pitch of the drone. How can I do this? How can the nose of the drone move to the desired position at a desired angle?