dronekit/dronekit-python #1223: can't connect my device that works on Ubuntu 20.04 to QGC version 4.2.9 via mavlink

hilaelb, 2024-06-26:
I have a simple code that I wrote on ubuntu 20.04. my goal is to connect to QGC and to see "Ready to fly" in the main screen.
my ip in the computer that qgc works on is 192.168.168.95 and my device ip is 192.168.168.5, when I run my device I can see in wireshark that there is communication between my device and qgc but still on the main screen it says "Disconnected"
this is my code on the device:

import rclpy
from rclpy.node import Node
from pymavlink import mavutil

class QGCcommNode(Node): 
    def _init_(self):
        super()._init_("qgc_comm")
        # Connect to the MAVLink instance
        self.master = mavutil.mavlink_connection('udpout:192.168.168.95:14570', source_system =1,source_component=1 )
        self.the_connection = mavutil.mavlink_connection('udpin:192.168.168.5:14575')
        self.timer = self.create_timer(1, self.send_heartbeat)
        

    # Function to send a heartbeat message
    def send_heartbeat(self):
        try:
            self.master.mav.heartbeat_send(
                type=mavutil.mavlink.MAV_TYPE_GENERIC,
                autopilot=mavutil.mavlink.MAV_AUTOPILOT_GENERIC,
                base_mode=0,
                custom_mode=0,
                system_status=mavutil.mavlink.MAV_STATE_ACTIVE
            )

            # Wait for a heartbeat
            msg = self.the_connection.recv_match(type='HEARTBEAT', blocking=True, timeout=1)
            if msg:
                self.get_logger().info("Heartbeat received from system")
            else:
                self.get_logger().warning("No heartbeat received")
        except Exception as e:
            self.get_logger().error(f"Error in send_heartbeat1: {str(e)}")

def main(args=None):
      
    rclpy.init(args=args)
    node = QGCcommNode() 
    rclpy.spin(node)
    rclpy.shutdown()   

when I run this code I get the "Heartbeat received from system" but yet the qgc is not connecting. in addition when I enter the Analyze Tools --> Mavlink Inspector this is the message that the qgc sends: 

[image]

I would love if someone wil help me :)
[image]

epkeefe123, 2024-06-26:
try connecting to it on port 5762 for the second port i had this struggle too

hilaelb, 2024-06-26:
> try connecting to it on port 1472 for the second port i had this struggle too its because the port is already taken by the script

what do you mean when you said "for the second port" ? where i need to change the port in QGC or in the device code?

epkeefe123, 2024-06-26:
so you are using dronekit right? you should just be able to put in the first four port numbers followed by a 2 so in your case it would be 14572 drone kit opens a second port from my experience you just need to connect to it first then it opens a second and third port with the numbers 1472 and 1473

epkeefe123, 2024-06-26:
i think

hilaelb, 2024-06-26:
i don't use dronekit- i only use in what i wrote in the code

hamishwillee, 2024-06-26:
So basically you're code is trying to pretend to be a flight stack and you want QGC to connect right? 

QGC probably expects a stream of heartbeats - that's what "connected" means.

hilaelb, 2024-06-27:
Exactly ! ok, so i will try what you suggested

hilaelb, 2024-06-27:
> So basically you're code is trying to pretend to be a flight stack and you want QGC to connect right?
> 
> QGC probably expects a stream of heartbeats - that's what "connected" means.

how do i sends a strem of HEARTBEAT? because i already sends a heartbeat message every second and still it does not work.

hamishwillee, 2024-06-27:
If you are already doing it this becomes a QGC question "what do you require for a connection to be established with a generic autopilot" You should ask in the QGC forums.

hilaelb, 2024-06-29:
> If you are already doing it this becomes a QGC question "what do you require for a connection to be established with a generic autopilot" You should ask in the QGC forums.

thank you