dronekit/dronekit-python #1224: take off without GPS

Amit5001, 2024-06-28:
Hi,
I'm trying to build autonomous drone for indoor. Currently my drone has micoair mtf-02 opticalflow and rngfnd calibrated.
when using only the controller the drone is able to maintain position.

My problems starts when i try to use dronekit- I want to take off in GUIDED_NOGPS or GUIDED.
in GUIDED (and LOITER too) I get an error- Need Position estimate so the drone don't arm.
In GUIDED_NOGPS the drone arm and start ascending but it gain inclination to one side and drift- it doesn't maintain its attitude.

1) Is there anything I need to change in configuration of the drone so it maintain attitude in GUIDED_NOGPS?
2) I read that the GUIDED mode problem related to the EKF need of home location. How can i set the EKF home using dronekit?

The script I used for takeoff:
[image]

Thanks!

GeraltRivskiy, 2024-10-17:
HI! Have you been able to solve your problem?
I have similar task, I need to implement simple goto functions in copter, but I can test it only indoors, where it cant obtain GPS signal. 
Am I understanding correct, that even positioning via velocity control needs GPS data or it could work fine with just accelerometer data?

hamishwillee, 2024-10-17:
These kinds of questions depend on your flight stack. I assume ArduPilot, but it might not be.

You would be best off asking on the autopilot discussion boards.

GeraltRivskiy, 2024-10-19:
@Amit5001 Thanks!