dronekit/dronekit-python #1227: Dronekit Python QuadCopter Kamikaze Drone Pitch Settings

Muhammetyilmaz7, 2024-08-13:
Hello, I am using dronekit python in quadcopter drone. This drone will perform a task autonomously. Let me briefly talk about the mission: The drone takes off and heads towards the specified location. After reaching a certain distance on the road, it must go to that location in a kamikaze dive manner. I am having trouble adjusting the pitch of the drone. How can I do this? How can the nose of the drone move to the desired position at a desired angle?

Muhammetyilmaz7, 2024-08-14:
I set the ANGLE_MAX and LOIT_ANG_MAX parameters to be at maximum angle. Now if I run the set_attitude_target.py script, can I get the result I want?

Muhammetyilmaz7, 2024-08-14:
When I ran the set_attitude_target.py script, the drone crashed. The outputs in Mav Console are as follows 
Got COMMAND_ACK: COMPONENT_ARM_DISARM: ACCEPTED AP: Arming motors ARMED AP: EKF3 IMU1 MAG0 in-flight yaw alignment complete AP: EKF3 IMU0 MAG0 in-flight yaw alignment complete Got COMMAND_ACK : DO_SET_MODE: ACCEPTED Mode LAND AP: EKF3 lane switch 1 AP: EKF primary changed:1 AP: Vibration compensation ON AP: GPS Glitch or Compass error AP: EKF3 IMU0 emergency yaw reset AP: EKF3 IMU1 emergency yaw reset Flight battery 100 percent AP : Crash: Disarming: AngErr=131>30, Accel=1.3<3.0 AP: Disarming motors AP: EKF primary changed:0 DISARMED AP: Glitch cleared AP: Vibration compensation OFF

Muhammetyilmaz7, 2024-08-14:
AP: EKF3 lane switch 1 AP: EKF primary changed:1 AP: Vibration compensation ON AP: EKF variance AP: GPS Glitch or Compass error AP: Crash: Disarming: AngErr=175>30, Accel=0.7<3.0 AP: Disarming motors AP: EKF primary changed:0 Flight battery 100 percent DISARMED AP: Glitch cleared AP: Vibration compensation OFF the drone crashes after giving these errors

peterbarker, 2024-08-14:
No.  Just... no.

closed 2024-08-14 by peterbarker