dronekit/dronekit-python #1229: Secure Mavlink

LDron, 2024-10-06:
Hello,
 I want to add an encryption algorithm to make the Mavlink protocol secure. Which files of Dronekit should I modify to add my algorithm?
Thanks

hamishwillee, 2024-10-06:
Generally people who want to make secure MAVLink just run MAVLink over a secure radio link. In any case dronekit is just a MAVLink abstraction - so you need to first decide what it is you're doing to encrypt MAVLink. There has been a lot of discussion about this over the years, I suggest you just run over a secure channel like everyone else.

closed 2024-10-06 by hamishwillee

LDron, 2024-10-10:
Thank you so mush @hamishwillee for replying. 
I want the following:
I use Dronekit with mavproxy to simulate a flying drone and I have an encryption algorithm. I want to all that algorithm to make Mavlink secure. In other words, when you capture the mavlink packets (say by wireshark) you can not read those messages because they are encrypted.

hamishwillee, 2024-10-10:
Same answer then. The simplest way to do this is to run over a secure radio so that no one can capture the packets. 

If you want to do it in the MAVLink layer this is possible but you will be incompatible with anything that does not implement this code - and MAVLink is unlikely to accept it, so you would have to fork. There is an example of how this might be done that has not been accepted into MAVLink in https://github.com/mavlink/mavlink/pull/1334