dronekit/dronekit-python #1234: Python Dronekit issue....Collections.MutableMapping

redden1388, 2025-01-16:
I am very new to this, however, I have done both pip dronekit installs.  But when I try to run any code I get a version of this.

  File "C:\Users\Finance\AppData\Local\Programs\Python\Python313\Lib\site-packages\dronekit\__init__.py", line 2689, in <module>
    class Parameters(collections.MutableMapping, HasObservers):

same error in PyCharm too, that's from VSCode. 

Seems the documents are a little load, missing and scrambled up.  Anyone have a solution or point me towards the resources for this project.

hamishwillee, 2025-01-19:
The project has to be built from source for relatively low version numbers of Python 3. There is no maintainer, so has little maintenance. Suggest you ask questions on the gitter channel or your specific autopilot discussion channels.

hamishwillee, 2025-01-19:
Docs are at https://dronekit.netlify.app/ - Yes, a bit out of date.

oc7o, 2025-02-02:
It's not ideal, but I just patched the issue at runtime, by inserting this before dronekit import:
```
import collections
import collections.abc
collections.MutableMapping = collections.abc.MutableMapping
```